TheImagingSource/tiscamera #496: Unable to get camera property name when using tcambin in pipeline

jdog816, 2022-11-14:
Have a DFM 22BUC03-ML camera and trying to read the gain and exposure settings, but I'm getting errors. This originally worked when calling "v4l2src name=camera" in the pipeline:

  gint fd;
  g_object_get(G_OBJECT(data->camera), "device-fd", &fd, NULL);

but now I get this error after switching from v4l2src to tcambin:

  GLib-GObject-WARNING **: 22:11:30.995: g_object_get_is_valid_property: object class 'GstTcamBin' has no property named 'device-fd'

I ran "tcam-ctrl -l" then "tcam-ctrl -c <SERIAL>" to see what the property names are, and I see "Gain" and "ExposureTime" listed, so I changed the code to this:

  gint gain, exposure;
  g_object_get(G_OBJECT(data->camera), "Gain", &gain, NULL);
  g_object_get(G_OBJECT(data->camera), "ExposureTime", &exposure, NULL);

but I still get similar errors:

  GLib-GObject-WARNING **: 22:04:03.173: g_object_get_is_valid_property: object class 'GstTcamBin' has no property named 'Gain'
  GLib-GObject-WARNING **: 22:04:02.991: g_object_get_is_valid_property: object class 'GstTcamBin' has no property named 'ExposureTime'

I searched the web for GstTcamBin but didn't find any info.

jdog816, 2022-11-14:
Here's the full chuck of relevant code:

     // Fetch gain and exposure settings
        
        gint fd;
        g_object_get(G_OBJECT(data->camera), "device-fd", &fd, NULL);
        
        struct v4l2_control ctrl;

        ctrl.id = V4L2_CID_GAIN;
        if ( ioctl(fd, VIDIOC_G_CTRL, &ctrl) == -1 ) {
            g_print ( "Unable to read gain from USB device fd= %d\n", fd );
            data->Gain = ctrl.value;
        }
        ctrl.id = V4L2_CID_EXPOSURE_ABSOLUTE;
        if ( ioctl(fd, VIDIOC_G_CTRL, &ctrl) == -1 ) {
            g_print ( "Unable to read exposure from USB device fd= %d\n", fd );
            data->Exposure = ctrl.value;
        }

It looks like more needs to be changed, now that v4l2src isn't being used. Do you have any recommendations on how to re-write it?

TIS-Stefan, 2022-11-15:
Hello

tcambin has a completely different property interface than v4l2. Therefore, we can use the same properties for v4l2, USB3Vision and GigEVision. 
You may look at the "set properties" sample at https://github.com/TheImagingSource/tiscamera/blob/master/examples/c/02-set-properties.c

I suppose, you use 
```
tcambin name=camera ! video/x.raw,format=BGRx,width=640,height=480,framerate=30/1
```
in your pipeline. There is no file descriptor. So you use 
```C
GError* err = NULL;
GstElement* source = gst_bin_get_by_name(GST_BIN(pipeline), "camera");
gst_element_set_state(pipeline, GST_STATE_READY);
tcam_property_provider_set_tcam_enumeration(TCAM_PROPERTY_PROVIDER(_source), "ExposureAuto", "Off", &err);
tcam_property_provider_set_tcam_enumeration(TCAM_PROPERTY_PROVIDER(_source), "GainAuto", "Off", &err);
tcam_property_provider_set_tcam_float(TCAM_PROPERTY_PROVIDER(_source), "ExposureTime", 30000.0, &err); 
tcam_property_provider_set_tcam_float(TCAM_PROPERTY_PROVIDER(_source), "Gain", 0.0, &err); 
```
 If you want to make it with error handling, then you add
```C
           if (err)
            {
                std::cout << _serial << "  " << err->message << '\n';
                g_error_free(err);
                err = NULL;
            }
```
after each property setting call.

closed 2022-11-15 by jdog816