TheImagingSource/tiscamera #498: USB3.0 vs USB3.1 gen1?

ShadeTechnik, 2022-11-27:
I don't know if this is a marketing mistake on the website or what but USB3.0 and USB3.1gen1 are the same thing. I can understand cameras that came before the USB3.1 revision was standardized being called USB3.0, however they are categorized separately and that gives the appearance that the cameras have different requirements. Only > USB3.1 gen2 should actually be separated as this denotes use of the higher speed 10gbps controller requirements.

Is there something I'm missing? I'm looking at the 12mp cameras over the 5mp I've been using and I just want to make sure that the USB revision spec difference is just semantics.

TIS-Kevin, 2022-11-28:
Dear ShadeTechnik,

we seperate between USB 3.0 and USB 3.1 because the USB 3.0 have the USB Micro B connector and USB 3.1 have USB Type C. So the cameras are not the "same".

The USB 3.0 (33U) cameras do the debayering to RGB24, YUY2 and Y411. The USB 3.1 cameras (37U) provide only raw data and debayering is done in the computer.

Additional: USB 3.0 has 12 bit for each colorchannel and USB 3.1 has 10 bit.

Likewise, the housing of the cameras are different.

Best Regards,
Kevin Herbrich

ShadeTechnik, 2022-11-28:
I didn't mean that the cameras are the "same", I meant USB 3.0 and USB 3.1gen1 are the "Same", because they are. I'm not sure if the C connector was available when the 3.0 standard was launched but all connectors are good up to 3.2 gen2 single lane. There technically is no 3.0 anymore, I believe there is actually no 3.1 either. It's all encompassed in 3.2 gen1, gen2, or gen2x2.

Point was that is a confusing way to label the cameras since I don't believe any of what you mentioned has anything to do with the USB version itself,  both 3.0 and 3.1g1 fall into the 5gbps host controller requirement. Labeling them something like "1st/2nd generation USB3", as in TIS's 1st and 2nd generation may have been more appropriate although even that could cause some confusion with the USB standard.

The debayering thing I was unaware of though, I havent gone through the 37 series documents yet as the possible requirement for higher res cameras just popped up. My application has been in limbo for quite some time and my memory of colorspace stuff is waning. I also don't recall personally why my program does what it does, just that there was a reason for it (I didn't write it). IIRC, I tell the cameras to output BGR3  and openCV converts it to RGB. I suppose this will have to change. What do the 37 series outputs look like to V4L2? I currently use 33UX264's

TIS-Stefan, 2022-11-28:
Helllo

You are absolutely right, USB naming is something for experts... USB 3.0 is currently the same as USB 3.1 Gen1 regarding speed. I guess, the "USB C" connector makes it rather "3.1 Gen 1.

The output of the camera is something GRRB and GRAY8, you can check with v4l2-ctl. OpenCVs cvtColor() can debayer the formats directly to RGB. (I currently have no own camera at hand, therefore, I point to v4l2-ctl for formats)

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar