TheImagingSource/tiscamera #499: Using multiple cameras on linux

dinhtuanan, 2022-12-01:
Hi,
I have 2 cameras: 

- DFK 33GX265    Serial: 31120204 aravis
- DFK 33GX264e  Serial: 39220008 aravis

I'm using Ubuntu 18.04.6 LTS

I plugged both of them through Ethernet Adapter Usb 3.0

Camera Info:

- [image]

- [image]

When I use tcam-capture application to open camera, I can only open 1 camera that plugged first and the other I can't open

- [image]

I would like to know how I can use two cameras or how to setting cameras right way

Did I miss something in config cameras?

Regards,
An

TIS-Stefan, 2022-12-01:
Hello

you may try to open the second camera with command
tcam-capture --reset.

Maybe that helps.

Stefan

dinhtuanan, 2022-12-01:
Hi Stefan, thank you so much for replying

I tried your suggestion but nothing happen. Did I do right?
[image]

TIS-Stefan, 2022-12-01:
Hello

you are right, that option was available in the precious version. But according to the documentation at https://www.theimagingsource.com/documentation/tiscamera/tcam-capture.html you can open the camera directly by serial:
tcam-capture --serial 31120204
and
tcam-capture --serial 39220008

I suppose, you can see the live stream of each of the cameras, if they are opened as only camera, right?

Stefan

TIS-Stefan, 2022-12-01:
Hello 

Just one addtion: You use a PI4, so please make sure to use low frame rates:
- avoid GigE Bandwidth exhaustion
- do not overwhelm the weak PI4 CPU
Stefan

dinhtuanan, 2022-12-01:
Thank you so much Stefan,

I will close this issue

Regards,
An

closed 2022-12-01 by dinhtuanan