TheImagingSource/tiscamera #500: TCAM_INSTALL_FORCE_PREFIX not honoured throughout?

chutsu, 2022-12-02:
Hello,

I'm trying to build and install `tiscamera` into my own defined install prefix, the option to do so is `TCAM_INSTALL_FORCE_PREFIX=ON` which works, however, not for all install targets (see screenshot below):

[image]

Looking at `tiscamera/CMakeInstall.cmake`, while the `CMAKE_INSTALL_PREFIX` are working for installing libraries, headers, udev rules ... when it came to installing gstreamer, gobject the `CMAKE_INSTALL_PREFIX` is no longer honoured. The [line 125 - 166](https://github.com/TheImagingSource/tiscamera/blob/master/CMakeInstall.cmake#L125) should be working but it isnt?

Thanks

TIS-Edgar, 2022-12-02:
Hi,

What does your cmake command look like?

Using:
> cmake -DCMAKE_INSTALL_PREFIX=/tmp -DTCAM_INSTALL_FORCE_PREFIX=ON ..

I get the expected output . 

>-- Build gstreamer-1.0 plugins:   ON
-- Support for GigE via aravis:   ON
-- Support for USB cameras:       ON
-- Support for LibUsb cameras:    ON
-- Build additional utilities:    ON
-- Build documentation            ON
-- Build tests                    OFF
-- 
-- 
-- Installation prefix:                 /tmp
-- 
-- Installing binaries to:              /tmp/bin
-- Installing libraries to:             /tmp/lib
-- Installing header to:                /tmp/include
-- Installing udev rules to:            /tmp/lib/udev/rules.d
-- Installing systemd units to:         /tmp/lib/systemd/system
-- Installing uvc-extensions to:        /tmp/share/theimagingsource/tiscamera/uvc-extension
-- Installing desktop files to:         /tmp/share/applications
-- Installing data files to:            /tmp/share/theimagingsource/tiscamera
-- Installing documentation to          /tmp/share/theimagingsource/tiscamera/doc
-- Installing gstreamer-1.0 to:         /tmp/lib/gstreamer-1.0
-- Installing gstreamer-1.0 header to:  /tmp/include/gstreamer-1.0
-- Installing gobject-introspection to: /tmp/share/gir-1.0
-- Installing introspection typelib to: /tmp/lib/girepository-1.0

Are you using a different version from current master(1.0.0)?

chutsu, 2022-12-02:
Hi I believe I'm using the latest commit `d9ada738`:

[image]

chutsu, 2022-12-02:
Checking out to `v-tiscamera-1.0.0` branch seems to yield the same results

TIS-Edgar, 2022-12-05:
Hmm, I am also on d9ada738 and I get the expected result.
Can delete the cmake cache before calling the cmake command?
Simply 'rm -rf ./*' in the build directory. Caching in cmake can be wonky at times....

chutsu, 2022-12-06:
Ah, thanks for the tip. That seems to have solved my problem 🤦🏻‍♂️. Sorry for the issue

closed 2022-12-06 by chutsu