TheImagingSource/tiscamera #505: DFG/USB2aud can not be detected on Ubuntu 18.04 with the kernel version 5.4

Jeffson-wu, 2023-01-19:
I followed the instructions to install the tiscamera.
I plugged in the DFG USB2aud. Then I used the following command to test:
**tcam-ctrl -l**
nothing output.

The result of **dmesg** after plugging in the board
[185591.759817] usb 1-2: new full-speed USB device number 8 using ohci-pci
[185592.266472] usb 1-2: config 1 interface 2 altsetting 0 endpoint 0x83 has invalid maxpacket 512, setting to 64
[185592.266475] usb 1-2: config 1 interface 3 altsetting 0 endpoint 0x84 has invalid maxpacket 512, setting to 64
[185592.266476] usb 1-2: config 1 interface 3 altsetting 3 endpoint 0x84 has invalid maxpacket 5060, setting to 1023
[185592.266477] usb 1-2: config 1 interface 3 altsetting 4 endpoint 0x84 has invalid maxpacket 2948, setting to 1023
[185592.266478] usb 1-2: config 1 interface 4 altsetting 0 endpoint 0x85 has invalid maxpacket 512, setting to 64
[185592.266479] usb 1-2: config 1 interface 4 altsetting 1 endpoint 0x5 has invalid maxpacket 512, setting to 64
[185592.266480] usb 1-2: config 1 interface 5 altsetting 0 endpoint 0x86 has invalid maxpacket 512, setting to 64
[185592.291840] usb 1-2: New USB device found, idVendor=199e, idProduct=8002, bcdDevice=40.01
[185592.291842] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[185592.291843] usb 1-2: Product: DFG/USB2pro
[185592.291843] usb 1-2: Manufacturer: The Imaging Source
[185592.291844] usb 1-2: SerialNumber: 01610409
[185592.314201] cx231xx 1-2:1.1: New device The Imaging Source DFG/USB2pro @ 12 Mbps (199e:8002) with 6 interfaces
[185592.322028] cx231xx 1-2:1.1: Identified as The Imaging Source DFG/USB2pro (card=25)
[185592.330306] i2c i2c-2: Added multiplexed i2c bus 4
[185592.330458] i2c i2c-2: Added multiplexed i2c bus 5
[185592.839050] cx25840 1-0044: cx23102 A/V decoder found @ 0x88 (cx231xx #0-0)
[185602.940097] cx25840 1-0044: loaded v4l-cx231xx-avcore-01.fw firmware (16382 bytes)
[185603.317708] cx231xx 1-2:1.1: v4l2 driver version 0.0.3
[185604.107581] cx231xx 1-2:1.1: Registered video device video0 [v4l2]
[185604.108251] cx231xx 1-2:1.1: Registered VBI device vbi0
[185604.109231] cx231xx 1-2:1.1: audio EndPoint Addr 0x83, Alternate settings: 3
[185604.109237] cx231xx 1-2:1.1: video EndPoint Addr 0x84, Alternate settings: 5
[185604.109242] cx231xx 1-2:1.1: VBI EndPoint Addr 0x85, Alternate settings: 2
[185604.109245] cx231xx 1-2:1.1: sliced CC EndPoint Addr 0x86, Alternate settings: 2

**lsusb result**
user@user:/dev$ lsusb -v -d 199e:8002

Bus 001 Device 008: ID 199e:8002 The Imaging Source Europe GmbH 
Couldn't open device, some information will be missing....

Jeffson-wu, 2023-01-20:
thanks kevin.
I will not use tiscamera then.

but I also tried with qv4l2 on ubuntu linux and can not get any stream.I am using DV-3030V DVD player as the input analog source.
On winodws 10, the device is working fine.do you know why then?
[image]

Jeffson-wu, 2023-01-20:
@TIS-Kevin 
also tried with this capture program, **select timeout error** occurs,please help,thanks

https://gist.github.com/maxlapshin/1253534

TIS-Stefan, 2023-01-20:
Hello

The DFG/USB2pro / aud is not supported by our tiscamera SDK . You approach with qvl2 is a good start.

Kernel 4.18 or 4.19 or above (not 4.20)  is needed. These kernel versions have the driver for DFG/USB2pro built in.

You may need to set the video norm and input channel properties are missing.  
But you can set these properties with v4l2-ctl (paket v4l-utils), before starting your application.

The possible input channel settings are
v4l2-ctl --set-input 0 // that is Compsite
v4l2-ctl --set-input 1 // that is S-VHS

The possible video norms, depending on your analogue camera are:
v4l2-ctl -s PAL-B
v4l2-ctl -s NTSC

This has to be done each time, you reconnect the device.

Stefan

Jeffson-wu, 2023-01-20:
@TIS-Stefan 
Thank you for the reply.

The ubuntu kernel version is 5.4 in my virtualbox and I checked the kernel output and lsusb command, I think the DFG/USB2pro driver already built in.

Also as I said earlier, I am not using analogue camera as the input source, instead I am using a DV-3030V DVD player as the input and I checked DFG/USB2pro is working well on windows with this DVD player.

Do you know what I should set for video norm and input channel properties for the DVD player before starting qv4l2 application?
I think this setting is ok:
**v4l2-ctl --set-input 0 // that is Compsite
v4l2-ctl -s PAL-B**

Thanks.

TIS-Stefan, 2023-01-23:
You should use the same settings in Linux, as you use them in Windows. Same input channel and same video norm.
qv4l2 should show something in my opinion. It is trial and error.

I am sorry for not having a better answer. The built in driver was not made by us.

Stefan

Jeffson-wu, 2023-01-23:
thanks Stefan

I think I made it work on Ubuntu desktop.

TIS-Stefan, 2023-01-23:
Hi Jeffson

Thank you for letting me know. I also recognized, you were in contact with the driver's author.

Stefan

Jeffson-wu, 2023-02-08:
Hi,Stefan

I want to reopen this ticket, I do made the device work on ubuntu desktop(with 640*480 res).But actually my goal is to make the device work on embedded device which have limited RAM.
I tried the DFG/USB2aud on my embedded device and it can only stream 400*360 maximum.The video will freeze if the resolution is higher than 400*360.
and I debugged the cx231xx driver and the result is cx231xx driver can not receive the data successfully from usb core if the resoultion is higher than 400*360.
**[  362.710000] cx231xx #0 print_err_status :URB packet 0, status -18 [Unknown].
[  362.710000] cx231xx #0 print_err_status :URB packet 1, status -18 [Unknown].
[  362.720000] cx231xx #0 print_err_status :URB packet 0, status -18 [Unknown].
[  362.730000] cx231xx #0 print_err_status :URB packet 0, status -18 [Unknown].
[  362.730000] cx231xx #0 print_err_status :URB packet 1, status -18 [Unknown].**

do you have any idea on this?
Thanks in advance.

TIS-Stefan, 2023-02-08:
Hello

I am very sorry, but I do not have the knowledge to help in this case.

Stefan

Jeffson-wu, 2023-02-08:
ok,thanks!

closed 2023-02-08 by Jeffson-wu