TheImagingSource/tiscamera #506: Tiscamerategra/tiscamsrc does not respect tcam-properties

stephenbaldwin, 2023-01-24:
Regardless of what I put into tiscamsrc tcam-properties, nothing changes. Also tiscamtegra documentation is missing from the website and doesn't seem to support tcam-properties

TIS-Stefan, 2023-01-25:
Stephen 
I am very sorry, but it would have been nice to give some context:
Please let me know:
* Computer model
* Camera model
* Steps to reproduce
* Bonus: Run "tcam-ctrl --version" and post its output. 
 
Thank you in advance!
Documentation can be found at https://www.theimagingsource.com/documentation/tiscamera/

Stefan

stephenbaldwin, 2023-01-25:
```
gst-launch-1.0 tcamsrc serial=$ID num-buffers=25 ! \
               video/x-bayer,format=grbg16,width=1920,height=1200,framerate=100/1 ! \
               tcamdutils tcam-properties=tcam,BalanceWhiteAuto=Off,BalanceWhiteBlue=3.9 ! \
               jpegenc ! \
               multifilesink location=./test-image-$ID.jpg max-files=1
```
does not alter the white balance.. You must use tcambin
```
gst-launch-1.0 tcambin serial=$ID tcam-properties=tcam,BalanceWhiteAuto=Off,BalanceWhiteBlue=3.9 ! \
                jpegenc ! \
                multifilesink location=./test-image-$ID.jpg max-files=1
```

However, tcambin does not expose ` num-buffers`

```
Model: DFM 36CR0234-ML Serial: 48220065 Type: tegra

Versions: 
	Tcam:	1.0.0_master/dd20eed7_rev_4005
	Aravis:	0.8_version_dd20eed7
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc_tests
```

stephenbaldwin, 2023-01-25:
Also https://www.theimagingsource.com/documentation/tcamtegrasrc/ is not viewable

TIS-Stefan, 2023-01-26:
Hello

you are right. It does not work with tcamsrc at all. Therefore, we need to write our own version of gst-launch, which allows us to set the properties are needed:

```Python
#!/usr/bin/env python3

import sys
import gi

gi.require_version("Tcam", "1.0")
gi.require_version("Gst", "1.0")

from gi.repository import Tcam, Gst, GLib

loop = GLib.MainLoop()

def bus_callback(bus, message, user_data):
    mtype = message.type

    if mtype == Gst.MessageType.EOS:
        # end-of-stream
        loop.quit()
    elif mtype == Gst.MessageType.ERROR:
        # Handle Errors
        err, debug = message.parse_error()
        print(err, debug)

        if err.message.startswith("Device lost ("):

            m = re.search('Device lost \((.*)\)', err.message)
            print("Device lost came from device with serial = {}".format(m.group(1)))

            # device lost handling should be initiated here
            # this example simply stops playback
            loop.quit()

    elif mtype == Gst.MessageType.WARNING:
        # Handle warnings
        err, debug = message.parse_warning()
        print(err, debug)

    return True

def main():
    Gst.init(sys.argv)  # init gstreamer

    serial = "45020091"

    pipeline =  Gst.parse_launch("tcambin name=bin"
                                 " ! video/x-raw,format=BGRx,width=1920,height=1200,framerate=100/1"
                                 " ! jpegenc ! multifilesink location=./test-image-{0}-%00d.jpg".format(serial) )

    # retrieve the bin element from the pipeline
    camera = pipeline.get_by_name("bin")
    
    # serial is defined, thus make the source open that device
    if serial is not None:
        camera.set_property("serial", serial)

    bus = pipeline.get_bus()
    bus.add_signal_watch()
    # bus.enable_sync_message_emission()
    bus.connect('message', bus_callback, None)        

    pipeline.set_state(Gst.State.READY)
    change_return, state, pending = pipeline.get_state(1000000000)

    tcamsrc = pipeline.get_by_name("tcambin-source")
    tcamsrc.set_property("num-buffers", 25)

    new_state = Gst.Structure.new_empty("tcam")

    new_state.set_value("BalanceWhiteAuto", "Off")
    new_state.set_value("BalanceWhiteBlue", 3.9)
    new_state.set_value("BalanceWhiteGreen", 1.0)
    new_state.set_value("BalanceWhiteRed", 1.0)

    camera.set_property("tcam-properties", new_state)

    pipeline.set_state(Gst.State.PLAYING)
    change_return, state, pending = pipeline.get_state(1000000000)

    loop.run()

    pipeline.set_state(Gst.State.NULL)

if __name__ == "__main__":
    main()
```

We use tcambin, get the tcamsource from tcambin, so we can set num-buffers to 25.
Also we set the white balance properties .

The bus callback is needed, because we want to end (EOS) after num-buffers images have been captured.
I also save all image, simply to see, whether I got num-buffers images.

I hope, this solves your issue.

Stefan

stephenbaldwin, 2023-02-09:
I'm having issues running this repeatedly where the cameras appear to still be in a playing state

stephenbaldwin, 2023-02-10:
also because tcambin-source is named implicitly under the hood I think this restricts me from building a pipeline that uses multiple tiscameras at once in a queue with customization

stephenbaldwin, 2023-02-10:
I think this is an active bug given documentation shows it should work: 
[image]

TIS-Edgar, 2023-02-10:
> also because tcambin-source is named implicitly under the hood I think this restricts me from building a pipeline that uses multiple tiscameras at once in a queue with customization

change 

    tcamsrc = pipeline.get_by_name("tcambin-source")
    
to

    tcamsrc = camera.get_by_name("tcambin-source")
    
and when you use the second bin use camera2 or the name you chose for that element.

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar