TheImagingSource/tiscamera #507: Cant open two cameras at same time when connected to different ethernet ports

Bardia-hub, 2023-02-01:
Hi i am having problems opening two cameras at same time when they are connected to different ethernet ports on linux 20.0.4

TIS-Kevin, 2023-02-02:
Dear Bardia-hub,

if you try to do that with tcam-capture it won't work.
Tcam-capture can only be used with 1 camera at the same time. (For each Window of Tcam-Capture).

Or do you have this problem with any sample we offer for using multiple cameras?

Please let us know!

Best Regards

Kevin Herbrich

mohammadhakim73, 2023-05-07:
Dear Bardia,

We had the same problem. The simplest solution is to set Gateway address of your cameras equal to IP Address of your Ethernet interfaces, and change IP of your cameras sth. in that range.

TIS-Stefan, 2023-05-08:
Dear Bardia

maybe the solution of Mohammad works. However, it would be interesting to see the output of
tcam-gigetool list

It lists all cameras which are in the same sub net.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar