TheImagingSource/tiscamera #508: get camera_time_ns outside of metadata?

Niv-R, 2023-02-14:
Is it possible to access camera_time_ns's value other than a frame's metadata?

I'd like to compensate for the camera's time drift in situations with infrequent frames where PTP is not available.

Niv-R, 2023-02-19:
Hi Kevin,

Can you please provide an example/snippet of how to do that? I couldn't find any example or documentation regarding  TimestampLatch/TimestampLatchValue

Niv-R, 2023-02-19:
Hi Kevin,

I tested TimestampLatchValue, but the returned value is signed int (32 bits) which easily overflows within ~2 seconds instead of unsigned 64 bit (like frame's metadata)

TIS-Kevin, 2023-02-20:
Dear Niv-R,

Which camera model did you use? 
Which Firmware Version is written in your camera?
Also which tiscamera version do you use?
Which program did you use?
Does it work in tcam-capture?

Please let me know

Best Regards,

Kevin

TIS-Kevin, 2023-02-20:
Dear Niv-R,

for TimestampLatch you have to work with tiscamera 1.0.
After that the property should be listed correctly.

Best Regards,

Kevin

Niv-R, 2023-05-07:
Updated tiscamera to version >=1.0. 
Confirmed TimestampLatchValue now works as expected.

closed 2023-05-07 by Niv-R