TheImagingSource/tiscamera #511: GStreamer tcamsrc Content is Black Image

engnfrc, 2023-02-23:
Hello,

I've installed tiscamera v1.0.0.4005 onto a system running Ubuntu 20.04 / Linux 5.4.0-137-lowlatency. The camera I'm using is a "DFM 25GX236-ML". When I run "IC Capture" from Windows I see images from the camera without any issues. On the Linux machine when I try to capture using GStreamer I'm just presented with a black image, I've tried both pipelines below and get the same results. I've also tried dropping resolutions and framerates. I've also included below the output from `tcam-ctrl -c 35710066`, which shows that I'm using valid GStreamer configurations. I also included some verbose output from GStreamer. Any ideas what the issue might be?

`gst-launch-1.0 -vv tcammainsrc serial=35710066 type=aravis ! video/x-bayer,format=gbrg,width=1920,height=1080,framerate=30/1 ! bayer2rgb ! videoconvert ! ximagesink`

`gst-launch-1.0 -vv tcammainsrc serial=35710066 type=aravis ! videoconvert ! ximagesink`

`tcam-ctrl -c 35710066`
> Available gstreamer-1.0 caps:
> video/x-raw,format=GRAY8,width=480,height=4,framerate={515625/21952,33/1,43/1,53/1,63/1,73/1,83/1,93/1,103/1,153/1,203/1,253/1,303/1,353/1,403/1,453/1,503/1,553/1,603/1,653/1,703/1,753/1,803/1,853/1,903/1,953/1,1003/1,1103/1,1203/1,1303/1,1403/1,1503/1,7425000/4753};
> video/x-raw,format=GRAY8,width=640,height=480,framerate={6875/3136,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,99000/679};
> video/x-raw,format=GRAY8,width=960,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,165000/1649};
> video/x-raw,format=GRAY8,width=1280,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,165000/1649};
> video/x-raw,format=GRAY8,width=1920,height=1080,framerate={1375/1344,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,46/1};
> video/x-raw,format=GRAY8,width=1920,height=1200,framerate={34375/37184,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,207/5};
> video/x-raw,format=GRAY8,width=[480,1920,16],height=[4,1200,4],framerate=[34375/37184,7425000/4753];
> video/x-raw,format=GRAY8,binning=1x2,width=[480,1920,16],height=[4,600,4],framerate=[34375/19264,7425000/4753];
> video/x-raw,format=GRAY8,binning=2x1,width=[480,960,16],height=[4,1200,4],framerate=[34375/37184,7425000/4753];
> video/x-raw,format=GRAY8,binning=2x2,width=[480,960,16],height=[4,600,4],framerate=[34375/19264,7425000/4753];
> video/x-raw,format=GRAY16_LE,width=480,height=4,framerate={515625/21952,33/1,43/1,53/1,63/1,73/1,83/1,93/1,103/1,153/1,203/1,253/1,303/1,353/1,403/1,453/1,503/1,553/1,603/1,653/1,703/1,753/1,3712500/4753};
> video/x-raw,format=GRAY16_LE,width=640,height=480,framerate={6875/3136,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,49500/679};
> video/x-raw,format=GRAY16_LE,width=960,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,82500/1649};
> video/x-raw,format=GRAY16_LE,width=1280,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,82500/1649};
> video/x-raw,format=GRAY16_LE,width=1920,height=1080,framerate={1375/1344,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,23/1};
> video/x-raw,format=GRAY16_LE,width=1920,height=1200,framerate={34375/37184,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,207/10};
> video/x-raw,format=GRAY16_LE,width=[480,1920,16],height=[4,1200,4],framerate=[34375/37184,3712500/4753];
> video/x-raw,format=GRAY16_LE,binning=1x2,width=[480,1920,16],height=[4,600,4],framerate=[34375/19264,3712500/4753];
> video/x-raw,format=GRAY16_LE,binning=2x1,width=[480,960,16],height=[4,1200,4],framerate=[34375/37184,3712500/4753];
> video/x-raw,format=GRAY16_LE,binning=2x2,width=[480,960,16],height=[4,600,4],framerate=[34375/19264,3712500/4753];
> video/x-bayer,format=gbrg,width=480,height=4,framerate={515625/21952,33/1,43/1,53/1,63/1,73/1,83/1,93/1,103/1,153/1,203/1,253/1,303/1,353/1,403/1,453/1,503/1,553/1,603/1,653/1,703/1,753/1,803/1,853/1,903/1,953/1,1003/1,1103/1,1203/1,1303/1,1403/1,1503/1,7425000/4753};
> video/x-bayer,format=gbrg,width=640,height=480,framerate={6875/3136,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,99000/679};
> video/x-bayer,format=gbrg,width=960,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,165000/1649};
> video/x-bayer,format=gbrg,width=1280,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,165000/1649};
> video/x-bayer,format=gbrg,width=1920,height=1080,framerate={1375/1344,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,46/1};
> video/x-bayer,format=gbrg,width=1920,height=1200,framerate={34375/37184,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,207/5};
> video/x-bayer,format=gbrg,width=[480,1920,16],height=[4,1200,4],framerate=[34375/37184,7425000/4753];
> video/x-bayer,format=gbrg,binning=1x2,width=[480,1920,16],height=[4,600,4],framerate=[34375/19264,7425000/4753];
> video/x-bayer,format=gbrg,binning=2x1,width=[480,960,16],height=[4,1200,4],framerate=[34375/37184,7425000/4753];
> video/x-bayer,format=gbrg,binning=2x2,width=[480,960,16],height=[4,600,4],framerate=[34375/19264,7425000/4753];
> video/x-bayer,format=gbrg16,width=480,height=4,framerate={515625/21952,33/1,43/1,53/1,63/1,73/1,83/1,93/1,103/1,153/1,203/1,253/1,303/1,353/1,403/1,453/1,503/1,553/1,603/1,653/1,703/1,753/1,3712500/4753};
> video/x-bayer,format=gbrg16,width=640,height=480,framerate={6875/3136,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,49500/679};
> video/x-bayer,format=gbrg16,width=960,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,82500/1649};
> video/x-bayer,format=gbrg16,width=1280,height=720,framerate={34375/22848,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,82500/1649};
> video/x-bayer,format=gbrg16,width=1920,height=1080,framerate={1375/1344,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,23/1};
> video/x-bayer,format=gbrg16,width=1920,height=1200,framerate={34375/37184,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,207/10};
> video/x-bayer,format=gbrg16,width=[480,1920,16],height=[4,1200,4],framerate=[34375/37184,3712500/4753];
> video/x-bayer,format=gbrg16,binning=1x2,width=[480,1920,16],height=[4,600,4],framerate=[34375/19264,3712500/4753];
> video/x-bayer,format=gbrg16,binning=2x1,width=[480,960,16],height=[4,1200,4],framerate=[34375/37184,3712500/4753];
> video/x-bayer,format=gbrg16,binning=2x2,width=[480,960,16],height=[4,600,4],framerate=[34375/19264,3712500/4753]

`GST_DEBUG=*:4 gst-launch-1.0 -e -vvv tcamsrc ! videoconvert ! ximagesink`

> 0:00:00.000160385 2837575 0x5606ea264000 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.16.3
> 0:00:00.000241363 2837575 0x5606ea264000 INFO                GST_INIT gst.c:587:init_pre: Using library installed in /usr/lib/x86_64-linux-gnu
> 0:00:00.000289231 2837575 0x5606ea264000 INFO                GST_INIT gst.c:605:init_pre: Linux hel-Nuvo 5.4.0-137-lowlatency #154-Ubuntu SMP PREEMPT Thu Jan 5 17:44:30 UTC 2023 x86_64
> 0:00:00.000731031 2837575 0x5606ea264000 INFO                GST_INIT gstmessage.c:128:_priv_gst_message_initialize: init messages
> 0:00:00.001777961 2837575 0x5606ea264000 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
[... cut ...]
> /GstPipeline:pipeline0/GstXImageSink:ximagesink0.GstPad:sink: caps = video/x-raw, width=(int)1920, height=(int)1200, framerate=(fraction)34375/37184, pixel-aspect-ratio=(fraction)1/1, format=(string)BGRx
> /GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, format=(string)GRAY16_LE, width=(int)1920, height=(int)1200, framerate=(fraction)34375/37184, pixel-aspect-ratio=(fraction)1/1
> /GstPipeline:pipeline0/GstTcamSrc:tcamsrc0.GstGhostPad:src.GstProxyPad:proxypad0: caps = video/x-raw, format=(string)GRAY16_LE, width=(int)1920, height=(int)1200, framerate=(fraction)34375/37184, pixel-aspect-ratio=(fraction)1/1

TIS-Stefan, 2023-02-23:
Hello

can you please run 
gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink

I do not expect a different result, but it is worth a test. 
Also please let me know which computer model you use.

engnfrc, 2023-02-23:
Same result. Just a smaller black image.

This system is a rugged embedded system, Neousys Nuvo-5000 (https://www.neousys-tech.com/en/product/feature/mezio/nuvo-5000)

`sudo lshw -short`
> H/W path           Device     Class          Description
>                               system         Nuvo-5000 (Skylake-S)
> /0                            bus            NVS-5000
> /0/0                          memory         128KiB BIOS
> /0/4                          processor      Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz
> /0/4/6                        memory         128KiB L1 cache
> /0/4/7                        memory         1MiB L2 cache
> /0/4/8                        memory         8MiB L3 cache
> /0/5                          memory         128KiB L1 cache
> /0/26                         memory         8GiB System Memory
> /0/26/0                       memory         4GiB SODIMM DDR4 Synchronous 2133 MHz (0.5 ns)
> /0/26/1                       memory         [empty]
> /0/26/2                       memory         4GiB SODIMM DDR4 Synchronous 2133 MHz (0.5 ns)
> /0/26/3                       memory         [empty]
> /0/120                        generic
> /0/100                        bridge         Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers
> /0/100/2                      display        HD Graphics 530
> /0/100/14                     bus            100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller
> /0/100/14/0        usb1       bus            xHCI Host Controller
> /0/100/14/0/5                 bus            USB 2.0 Hub
> /0/100/14/0/5/2               multimedia     HD USB Camera
> /0/100/14/0/5/3               multimedia     HD USB Camera
> /0/100/14/0/5/4               multimedia     HD USB Camera
> /0/100/14/0/6                 bus            USB 2.0 Hub
> /0/100/14/0/6/1               generic        Microchip Vendor Example Device
> /0/100/14/0/6/4               communication  u-blox 7 - GPS/GNSS Receiver
> /0/100/14/0/a                 bus            HighSpeed Hub
> /0/100/14/0/a/1               multimedia     H264 USB Camera
> /0/100/14/0/a/2               multimedia     H264 USB Camera
> /0/100/14/0/a/3               multimedia     H264 USB Camera
> /0/100/14/0/c                 bus            HighSpeed Hub
> /0/100/14/0/c/1               input          Dell KB216 Wired Keyboard
> /0/100/14/0/c/2               input          USB Optical Mouse
> /0/100/14/1        usb2       bus            xHCI Host Controller
> /0/100/16                     communication  100 Series/C230 Series Chipset Family MEI Controller #1
> /0/100/16.3                   communication  100 Series/C230 Series Chipset Family KT Redirection
> /0/100/17          scsi3      storage        Q170/Q150/B150/H170/H110/Z170/CM236 Chipset SATA Controller [AHCI Mode]
> /0/100/17/0.0.0    /dev/sda   disk           1TB Samsung SSD 850
> /0/100/17/0.0.0/1  /dev/sda1  volume         31MiB BIOS Boot partition
> /0/100/17/0.0.0/2  /dev/sda2  volume         40GiB EXT4 volume
> /0/100/17/0.0.0/3  /dev/sda3  volume         891GiB EXT4 volume
> /0/100/1c                     bridge         100 Series/C230 Series Chipset Family PCI Express Root Port #3
> /0/100/1c/0        enp1s0     network        I210 Gigabit Network Connection
> /0/100/1c.4                   bridge         100 Series/C230 Series Chipset Family PCI Express Root Port #5
> /0/100/1c.4/0      enp2s0     network        I210 Gigabit Network Connection
> /0/100/1c.5                   bridge         100 Series/C230 Series Chipset Family PCI Express Root Port #6
> /0/100/1c.5/0      enp3s0     network        I210 Gigabit Network Connection
> /0/100/1c.6                   bridge         100 Series/C230 Series Chipset Family PCI Express Root Port #7
> /0/100/1c.6/0      enp4s0     network        I210 Gigabit Network Connection
> /0/100/1c.7                   bridge         100 Series/C230 Series Chipset Family PCI Express Root Port #8
> /0/100/1c.7/0      enp5s0     network        I210 Gigabit Network Connection
> /0/100/1f                     bridge         Q170 Chipset LPC/eSPI Controller
> /0/100/1f.2                   memory         Memory controller
> /0/100/1f.3                   multimedia     100 Series/C230 Series Chipset Family HD Audio Controller
> /0/100/1f.4                   bus            100 Series/C230 Series Chipset Family SMBus
> /0/100/1f.6        enp0s31f6  network        Ethernet Connection (2) I219-LM
> /0/1                          system         PnP device PNP0c02
> /0/2                          system         PnP device PNP0c02
> /0/3                          system         PnP device PNP0c02
> /0/6                          system         PnP device PNP0c02
> /0/7                          system         PnP device PNP0b00
> /0/8                          generic        PnP device INT3f0d
> /0/9                          input          PnP device PNP0303
> /0/a                          communication  PnP device PNP0501
> /0/b                          communication  PnP device PNP0501
> /0/c                          communication  PnP device PNP0501
> /0/d                          communication  PnP device PNP0501
> /0/e                          system         PnP device PNP0c02
> /0/f                          system         PnP device PNP0c02
> /1                            power          CRB Battery 0
> /2                            power          OEM Define 5
> /3                 usb3       bus            USB/IP Virtual Host Controller
> /3/1                          multimedia     USB 2.0 Camera
> /3/2                          multimedia     USB 2.0 Camera
> /3/3                          multimedia     USB 2.0 Camera
> /4                 usb4       bus            USB/IP Virtual Host Controller

engnfrc, 2023-02-23:
I figured it out, my NIC was defaulting to an mtu of 1500 and I changed it to 9000 using: `sudo ip link set dev enp0s31f6 mtu 9000`. Searching through the documentation I see there's a brief mention of this setting on the Troubleshooting page (https://www.theimagingsource.com/documentation/tiscamera/troubleshooting.html?highlight=mtu).

TIS-Stefan, 2023-02-24:
Thank you for letting us know.

closed 2023-02-24 by TIS-Stefan