TheImagingSource/tiscamera #512: Pixel Artefact during hardware trigger mode [DFM 22BUC03-ML]

chutsu, 2023-02-28:
**Describe the bug**

I have a DFM 22BUC03-ML connected to my laptop. I'm experimenting with hardware triggering the camera, I have done this by programming a Arduino to set a pin HIGH and LOW periodically within the camera's frame-rate. What I have noticed when I inspect the images via `tcam-capture` is that during *hardware-trigger* mode the image contains pixel-artefacts as if there are dead pixels in the image, however as soon as I switch back to *software-trigger* by going to `Properties > Trigger > Trigger Mode > Off`, the pixel artefacts disappears.

I have tried:
- Changing image resolution
- Hardware triggering different rates
- Color to GRAY8
to no avail.

Screenshot of `tcam-capture` in hardware trigger mode, I have the camera lens cap on for illustration purposes only. From the screenshot you can see a sprinkle of red / blue pixels. This disappears as soon as I turn trigger mode to software trigger.
[image]

**Hardware Trigger with Pixel Artefacts**
[image]

**Software Trigger without Pixel Artefacts**
[image]

**computer used (please complete the following information):**
 - OS: Ubuntu 20.04
 - Architecture/Platform: amd64, ThinkPad P52
 - Camera  model: DFM 22BUC03-ML
 - tiscamera version: 1.0.0_HEAD/d9ada738_rev_4011

TIS-Stefan, 2023-02-28:
Hello 

I am afraid, this is caused by the camera and can not be fixed. I am aware, that is a non helpful answer and I am a little bit ashamed. 

Stefan

chutsu, 2023-02-28:
To clarify, is this caused by the sensor or some internal hardware / software issue? To say I'm dissapointed would be an under-statement as I plan on doing camera calibration with these sensors ... not entirely sure how to proceed with these pixel artifacts.

chutsu, 2023-02-28:
```
19220362
19220363
```
🙏🏻

TIS-Stefan, 2023-02-28:
Hello

My firmware team answered. They say, it is a known sensor issue, that can not be changed. 

Stefan

chutsu, 2023-03-02:
I purchased the cameras from a distributor in the UK almost over a year ago. Is there a chance I can return based on the fact these cameras do not meet my requirements. Hardware triggering is the primary reason why I purchased theses industrial cameras in the first place and I feel duped into purchasing these cameras with a known sensor issue...

TIS-Stefan, 2023-03-02:
Please ask your distributor. But waiting over a year for seeing this issue is a little bit long. 

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.