TheImagingSource/tiscamera #514: Camera getting disconnected after run gstreamer pipeline code

KushalBiyani, 2023-03-13:
**Issue**
We have created a docker image with Ubuntu 20.04 with the commands specified in the [Building tiscamera](https://github.com/TheImagingSource/tiscamera#:~:text=dependency%2Dmanager%20install-,Building%20tiscamera,-The%20following%20commands)  on GitHub.
We were able to list down the camera using **tcam-ctrl -l** command.

[image]

When we are starting the pipeline using below command : 
**gst-launch-1.0 tcambin ! videoconvert ! ximagesink**

we are getting the below error : 

[image]

And while running this [code](https://www.theimagingsource.com/en-us/documentation/tiscamera/tutorial.html#receiving-images) our camera is getting disconnected every time and we are not able to receive any frames and not able to list down the camera also.

TIS-Edgar, 2023-03-14:
Hi,

Could you set the following environment variable:

```
export GST_DEBUG=ximagesink:5,tcam-libtcam:5
```

This will enable debug output for our backend and the ximagesink.

Does tcam-capture work?
Does the xvimagesink work?
Does tiscamera work outside of the docker container?

The disconnects should also be logged in the debug output.
Did you adjust the MTU to 9000?

KushalBiyani, 2023-03-15:
**Tiscamera is running outside the docker container**. We are using docker desktop on windows machine form where we are able to receive the feeds on IC capture software.

While running the **tcam-ctrl -l** command we are getting this output: 

[image]

On running **gst-launch-1.0 tcambin ! videoconvert ! ximagesink** we were getting this output: 

[image]

On running **gst-launch-1.0 tcambin ! videoconvert ! xvimagesink** we were getting this output: 

[image]

On running **tcam-capture** we were getting this output: 

[image]

On running the code from the [app-sink.py](https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/07-appsink.py) file in Github example we get this error and camera also got disconnected : 

[image]

TIS-Edgar, 2023-03-16:
We are trying to find the cause of this issue.
Docker on Windows makes things more "interesting" than needed.

Can you set the following environment variable
```export ARV_DEBUG=all:3```

and capture the output of wireshark for us?
You should have your native interface available and a WSL interface.
Output of both would be of interest at this point.

Please disable host camera services before doing this.

You can send the capture to support@theimagingsource.com referencing this issue so that potential confidential data will be handled as such.

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar