TheImagingSource/tiscamera #515: Receive Image in Hardware trigger mode

dinhtuanan, 2023-03-16:
Hi, I use 2 cameras to capture image when trigger them. I would like to know if I can limit the image a camera gives me right at the time of trigger. Because when i trigger it sometimes i get 2 pictures of the same camera instead of 2 pictures from 2 different cameras.

If you have examples of this or instruction, could you suggest them to me?

Regards,
An

dinhtuanan, 2023-03-16:
My callback function to get images from 2 cameras
[image]

TIS-Stefan, 2023-03-16:
Hello

In case you use an external hardware trigger, make sure, you have a clear, not noisy trigger signal. Also make sure, the current of your trigger sure has at leat 3.3V.

If you use a mechanical switch for the trigger, please make sure that it is debounced.

Stefan

dinhtuanan, 2023-03-17:
Hello Stefan

Can "Trigger Mask Time" be dynamic time? Because I want After 2 pictures have been taken, I want to execute doSomething() function to allow trigger to continue taking pictures.

Regards,
An

TIS-Stefan, 2023-03-17:
Hi An

I am very sorry guessing... but I even to not know, which camera you use. Also I do not know, how the trigger signal is generated.
Without getting all necessary information, I guess into the blue and give general answer. I am aware, that is wasting of time.

I do not know your device, therefore, I do not know, how it generates trigger signals. Do you have a link to a specification of this device?

Therefore, please clarify:

Do you get exactly one image from a camera at one trigger signal?

Also you showed the callback function only. The interesting part, how you setup the camera and use the callback is not shown. 

Are you sure, you get all images? It could be possible, you miss one image of one camera. But for I guess again, please let me know, which camera model you use.
And how it is connected  to the computer.
And which computer model you use.

Stefan

dinhtuanan, 2023-03-17:
Hi Stefan,

I'm so sorry for missing information. I will close this issue and open new when I prepare full information

Regards,
An

closed 2023-03-17 by dinhtuanan