TheImagingSource/tiscamera #517: Reading Whitebalance values after One-Push Whitebalance

elRadish, 2023-04-21:
I'm still on TisCamera 0.14, Linux 18.04 on Jetson Nano.

I’m trying to do a one-push autofocus and read back the White Balance values for red, green and blue to save them permanently.

But I can’t figure out how to do this. 
In IC Capture on Windows, the current values are displayed in the device properties, but when I read them back with `Tis.Get_Property("Whitebalance Blue").value`, it doesn’t change, only, when I set it manually.

Is this possible with TisCamera 0.14?

TIS-Stefan, 2023-04-21:
Hello

Are you sure, you mean one push auto-focus, not one push Whitebalance?

Stefan

elRadish, 2023-04-22:
Oops, no, of course one-push-whitebalance. Thanks!

TIS-Stefan, 2023-04-24:
If you run the white balance one push in your Linux software, does the live video change to white balanced images?

(I ask so many questions, because we ended the support for tiscamera 0.14, since we use 1.1 by now.)

Stefan

elRadish, 2023-04-28:
Yes, it does, but the Properties keep the same. 
The same happens in tcam-capture - it doesn't update the manual values which would be pretty useful:
My guess is that the values are read from the image metadata instead of the camera properties?

I'll try to migrate to 1.1 asap!

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar