TheImagingSource/tiscamera #519: Stuck setting pipeline state to ready - pipeline.set_state(Gst.State.READY)

Niv-R, 2023-05-22:
**Describe the bug**
After upgrading to to TIS 1.1.1 from 0.14 I started experiencing failures (stuck) when try to change the pipeline's state.
Admittedly sometimes my program closes abruptly without setting state to NULL

My pipeline: 
```
tcamsrc serial=... name=source ! video/x-raw,format=GRAY12p,width=1920,height=1080,framerate=8192/225 ! valve name=valve ! appsink name=sink
```

Python code:
```
pipeline: Gst.Pipeline = Gst.parse_launch(self.pipeline_str)
pipeline.set_state(Gst.State.NULL)  # <-- seems redundent but it did help in the past
pipeline.set_state(Gst.State.READY) # <-- this often gets stuck
...
pipeline.set_state(Gst.State.PLAYING)
```

**computer used (please complete the following information):**
 - OS: Ubuntu 20.04
 - Architecture/Platform: amd64 Desktop
 - Camera  model:DMK 39GX265-Z20-TI
 - tiscamera version:
	  Tcam:	1.1.1_master/ee45fcf7_rev_4142
	  Aravis:	0.8_version_e977fa4
	  Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc_tests

**Additional context**
I've set `Gst.debug_set_default_threshold(Gst.DebugLevel.INFO)` and produced the following:
```
0:01:05.041644387   321 0x7f31cc007660 INFO            GST_PIPELINE gstparse.c:336:gst_parse_launch_full: parsing pipeline description 'tcamsrc serial=... name=source ! video/x-raw,format=GRAY12p,width=1920,height=1080,framerate=8192/225 ! valve name=valve ! appsink name=sink'
0:01:05.041678006   321 0x7f31cc007660 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "tcamsrc"
0:01:05.041799441   321 0x7f31cc007660 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "tcammainsrcdeviceprovider"
0:01:05.041851484   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstTcamSrc@0x7f31b003b280> adding pad 'src'
0:01:05.041874977   321 0x7f31cc007660 INFO              GST_STATES gstbin.c:2090:gst_bin_get_state_func:<tcamsrc1> getting state
0:01:05.042708181   321 0x7f31cc007660 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so" loaded
0:01:05.042725149   321 0x7f31cc007660 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "valve"
0:01:05.042772372   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstValve@0x6ee0d30> adding pad 'src'
0:01:05.042796402   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstValve@0x6ee0d30> adding pad 'sink'
0:01:05.042819030   321 0x7f31cc007660 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "appsink"
0:01:05.042834023   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstBaseSink@0x7f31b004f7d0> adding pad 'sink'
0:01:05.042858178   321 0x7f31cc007660 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "pipeline"
0:01:05.042919233   321 0x7f31cc007660 INFO            GST_PIPELINE grammar.y:648:gst_parse_perform_link: linking some pad of GstTcamSrc named source to some pad of GstValve named valve (0/0) with caps "video/x-raw, format=(string)GRAY12p, width=(int)1920, height=(int)1080, framerate=(fraction)8192/225"
0:01:05.042934565   321 0x7f31cc007660 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "capsfilter"
0:01:05.043003256   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstBaseTransform@0x7f31b0052250> adding pad 'sink'
0:01:05.043025372   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstBaseTransform@0x7f31b0052250> adding pad 'src'
0:01:05.043039571   321 0x7f31cc007660 INFO              GST_STATES gstbin.c:2090:gst_bin_get_state_func:<pipeline1> getting state
0:01:05.043057961   321 0x7f31cc007660 INFO              GST_STATES gstelement.c:2688:gst_element_continue_state:<capsfilter0> completed state change to NULL
0:01:05.043070028   321 0x7f31cc007660 INFO               GST_EVENT gstevent.c:1579:gst_event_new_reconfigure: creating reconfigure event
0:01:05.043090516   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstutils.c:1771:gst_element_link_pads_full: trying to link element source:(any) to element capsfilter0:sink
0:01:05.043101518   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:928:gst_element_get_static_pad: found pad capsfilter0:sink
0:01:05.043109460   321 0x7f31cc007660 INFO                GST_PADS gstutils.c:1587:prepare_link_maybe_ghosting: source and capsfilter0 in same bin, no need for ghost pads
0:01:05.043121354   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:2377:gst_pad_link_prepare: trying to link source:src and capsfilter0:sink
0:01:05.043133885   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:4237:gst_pad_peer_query:<src:proxypad1> pad has no peer
0:01:05.043148997   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:4237:gst_pad_peer_query:<capsfilter0:src> pad has no peer
0:01:05.043164930   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:2585:gst_pad_link_full: linked source:src and capsfilter0:sink, successful
0:01:05.043173016   321 0x7f31cc007660 INFO               GST_EVENT gstevent.c:1579:gst_event_new_reconfigure: creating reconfigure event
0:01:05.043180670   321 0x7f31cc007660 INFO               GST_EVENT gstpad.c:5812:gst_pad_send_event_unchecked:<source:src> Received event on flushing pad. Discarding
0:01:05.043192650   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstutils.c:1771:gst_element_link_pads_full: trying to link element capsfilter0:src to element valve:(any)
0:01:05.043201698   321 0x7f31cc007660 INFO        GST_ELEMENT_PADS gstelement.c:928:gst_element_get_static_pad: found pad capsfilter0:src
0:01:05.043213072   321 0x7f31cc007660 INFO                GST_PADS gstutils.c:1034:gst_pad_check_link: trying to link capsfilter0:src and valve:sink
0:01:05.043227247   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:4237:gst_pad_peer_query:<src:proxypad1> pad has no peer
0:01:05.043243201   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:4237:gst_pad_peer_query:<valve:src> pad has no peer
0:01:05.043254085   321 0x7f31cc007660 INFO                GST_PADS gstutils.c:1587:prepare_link_maybe_ghosting: capsfilter0 and valve in same bin, no need for ghost pads
0:01:05.043264643   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:2377:gst_pad_link_prepare: trying to link capsfilter0:src and valve:sink
0:01:05.043275364   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:4237:gst_pad_peer_query:<src:proxypad1> pad has no peer
0:01:05.043289304   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:4237:gst_pad_peer_query:<valve:src> pad has no peer
0:01:05.043303232   321 0x7f31cc007660 INFO                GST_PADS gstpad.c:2585:gst_pad_link_full: linked capsfilter0:src and valve:sink, successful
0:01:05.043311410   321 0x7f31cc007660 INFO               GST_EVENT gstevent.c:1579:gst_event_new_reconfigure: creating reconfigure event
0:01:05.043318691   321 0x7f31cc007660 INFO               GST_EVENT gstpad.c:5812:gst_pad_send_event_unchecked:<capsfilter0:src> Received event on flushing pad. Discarding
[... 35 lines cut ...]
0:01:05.043944364   321 0x7f31f4007c10 INFO              GST_STATES gstbin.c:2621:gst_bin_element_set_state:<capsfilter0> skipping transition from NULL to  NULL
0:01:05.043954247   321 0x7f31f4007c10 INFO              GST_STATES gstbin.c:2952:gst_bin_change_state_func:<pipeline1> child 'capsfilter0' changed state to 1(NULL) successfully
0:01:05.043966997   321 0x7f31f4007c10 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<source> current NULL pending VOID_PENDING, desired next NULL
0:01:05.043984196   321 0x7f31f4007c10 INFO              GST_STATES gstelement.c:2688:gst_element_continue_state:<source> completed state change to NULL
0:01:05.043996007   321 0x7f31f4007c10 INFO              GST_STATES gstbin.c:2952:gst_bin_change_state_func:<pipeline1> child 'source' changed state to 1(NULL) successfully
0:01:05.044005844   321 0x7f31f4007c10 INFO              GST_STATES gstelement.c:2688:gst_element_continue_state:<pipeline1> completed state change to NULL

# pipeline.set_state(Gst.State.READY) 

[... cut ...]
0:01:05.044365861   321 0x7f31f4008950 INFO              GST_STATES gstbin.c:2952:gst_bin_change_state_func:<pipeline1> child 'capsfilter0' changed state to 2(READY) successfully
0:01:05.044376662   321 0x7f31f4008950 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<source> current NULL pending VOID_PENDING, desired next READY
0:01:05.044455946   321 0x7f31f4008950 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "tcammainsrc"
0:01:05.044548260   321 0x7f31f4008950 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstBaseSrc@0x7f31f4010310> adding pad 'src'
```

Niv-R, 2023-05-30:
Complete logs with GST_DEBUG set (this time without the loop)

```
0:00:00.000065044    15      0x7f05e30 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.16.3
0:00:00.000080513    15      0x7f05e30 INFO                GST_INIT gst.c:587:init_pre: Using library installed in /usr/lib/x86_64-linux-gnu
0:00:00.000097969    15      0x7f05e30 INFO                GST_INIT gst.c:605:init_pre: Linux XXX 5.19.0-41-generic #42~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC XXX 2 x86_64
0:00:00.000241265    15      0x7f05e30 INFO                GST_INIT gstmessage.c:128:_priv_gst_message_initialize: init messages
0:00:00.000384767    15      0x7f05e30 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.000502596    15      0x7f05e30 INFO      GST_PLUGIN_LOADING gstplugin.c:318:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.000560351    15      0x7f05e30 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.000571559    15      0x7f05e30 INFO      GST_PLUGIN_LOADING gstplugin.c:228:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.000581768    15      0x7f05e30 INFO            GST_REGISTRY gstregistry.c:1733:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.009590088    15      0x7f05e30 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.x86_64.bin in 0.008997 seconds
0:00:00.009634164    15      0x7f05e30 INFO            GST_REGISTRY gstregistry.c:1592:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.011301052    15      0x7f05e30 INFO            GST_REGISTRY gstregistry.c:1691:scan_and_update_registry: Registry cache has not changed
0:00:00.011311721    15      0x7f05e30 INFO            GST_REGISTRY gstregistry.c:1768:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.011320012    15      0x7f05e30 INFO                GST_INIT gst.c:806:init_post: GLib runtime version: 2.64.6
0:00:00.011328314    15      0x7f05e30 INFO                GST_INIT gst.c:808:init_post: GLib headers version: 2.64.6
0:00:00.011335211    15      0x7f05e30 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:01:00.486431178    15 0x7f7e0000ba40 INFO            tcam-libtcam libtcam_base.cpp:63:print_setup_to_logger:
The following library versions are used:
	Tcam:	1.1.1_master/ee45fcf7_rev_4142
	Aravis:	0.8_version_e977fa4
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc_tests
0:01:00.486448626    15 0x7f7e0000ba40 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsttcamsrc.so" loaded
0:01:00.486458711    15 0x7f7e0000ba40 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "tcammainsrcdeviceprovider"
0:01:00.486623816    15 0x7f7e0000d4c0 INFO            tcam-libtcam aravis_utils.cpp:187:fetch_gige_daemon_device_list: Failed to create shmkey. Not using gige-daemon to enumerate devices.
0:01:01.542435037    15 0x7f7e0000ba40 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "tcammainsrcdeviceprovider"
0:01:16.735388159   336 0x7f7df4004270 INFO            GST_PIPELINE gstparse.c:336:gst_parse_launch_full: parsing pipeline description 'tcamsrc serial=123 name=source ! video/x-raw,format=GRAY12p,width=1920,height=1080,framerate=8192/225 ! valve name=valve ! appsink name=sink'
0:01:16.735442083   336 0x7f7df4004270 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "tcamsrc"
0:01:16.736115527   336 0x7f7df4004270 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "tcammainsrcdeviceprovider"
0:01:16.736270458   336 0x7f7df4004270 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstTcamSrc@0x7f7e1803b0e0> adding pad 'src'
0:01:16.736312768   336 0x7f7df4004270 INFO              GST_STATES gstbin.c:2090:gst_bin_get_state_func:<tcamsrc0> getting state
0:01:16.738457011   336 0x7f7df4004270 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so" loaded
0:01:16.738473165   336 0x7f7df4004270 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "valve"
0:01:16.738542192   336 0x7f7df4004270 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstValve@0x8800d30> adding pad 'src'
0:01:16.738573361   336 0x7f7df4004270 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstValve@0x8800d30> adding pad 'sink'
0:01:16.738923921   336 0x7f7df4004270 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstapp.so" loaded
0:01:16.738939324   336 0x7f7df4004270 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "appsink"
[... cut ...]
0:01:16.741150283   336 0x7f7e18052990 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<source> current NULL pending VOID_PENDING, desired next READY
0:01:16.741248339   336 0x7f7e18052990 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:363:gst_element_factory_create: creating element "tcammainsrc"
0:01:16.741351822   336 0x7f7e18052990 INFO        GST_ELEMENT_PADS gstelement.c:671:gst_element_add_pad:<GstBaseSrc@0x7f7e4000e310> adding pad 'src'
0:01:16.741405606   336 0x7f7e18052990 DEBUG            tcammainsrc mainsrc_device_state.cpp:326:open_camera:<source> Trying to open device with serial='123 and type='aravis'.
```

TIS-Stefan, 2023-05-30:
Hello
why you have a camera with serial number "123"?

Stefan

Niv-R, 2023-05-30:
it's not 123, just changed the logs, used `...` earlier

TIS-Stefan, 2023-05-31:
I was concerned, because the 123 is the serial number of the cameras, if they have no firmware installed. 
I now assume, the error happens in the docker container only. Did you pass the --net host parameter when starting the container?

Niv-R, 2023-05-31:
>Did you pass the --net host parameter when starting the container?

Yes

>Can you please share the docker script?

Here's a simplified `docker-compose.yml` example 
you'll need to install `tiscamera_1.1.1.4142_amd64_ubuntu_2004.deb` inside of it

```
version: "3.9"
services:
  test-camera:
    privileged: true
    image: ubuntu:focal
    volumes:
      - /etc/network:/etc/network
    network_mode: host
```

TIS-Stefan, 2023-05-31:
This is my working Dockerfile on Ubuntu 22.04:
```
FROM ubuntu
ARG DEBIAN_FRONTEND=noninteractive
RUN apt-get -y update
RUN apt-get -y install libgstreamer1.0-0
RUN apt-get -y install gstreamer1.0-tools
COPY ./tiscamera_1.0.0.4005_amd64_ubuntu_1804.deb /
RUN apt install -y /tiscamera_1.0.0.4005_amd64_ubuntu_1804.deb
COPY ./tcamdutils_1.0.0.560_amd64.deb /
RUN apt install -y /tcamdutils_1.0.0.560_amd64.deb
RUN groupadd -g 1000 tcam
RUN useradd -d /home/tcam -s /bin/bash -m tcam -u 1000 -g 1000
USER tcam
ENV HOME /home/tcam
CMD /usr/bin/tcam-capture
```
It is built with
```
sudo docker build -t tiscamera .
```
(Do not forget the "." at the end.) You may need to change the tiscamera version numbers.

It start this with
```
sudo docker run -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY -h $HOSTNAME -v $HOME/.Xauthority:/home/tcam/.Xauthority --net host --privileged --volume=/dev:/dev:rw tiscamera
```
tcam-capture is started and I get a video from my GigE camera. 

Stefan

TIS-Stefan, 2023-05-31:
Hello 

> Alright, any insight regarding the pipeline getting stuck?

I am sorry, no But it could be something, with the user group stuff. You may add that to your Dockerfile and build the container again.

Stefan

Niv-R, 2023-07-05:
Starting `tcam-gige-daemon` on docker start before accessing the camera seems to have solved this issue for me.

closed 2023-07-05 by Niv-R