TheImagingSource/tiscamera #523: Cannot get image from DFK39GX265-Z20

ShadeTechnik, 2023-05-29:
I have this camera on evaluation and haven't quite been able to get any images from it. I'm using Manjaro but I also briefly tried it with Debian, On the Manjaro box, I compiled Tiscamera from source. I can connect to the camera and it shows up with tcam-gigetool but that's about it. With tcam-capture it doesn't  list the camera. I tried using arv-viewer and that does list the camera but it just glitches out when I try to open it, that may be a problem with arv-viewer itself though. I'm connecting directly to the PCs Ethernet port and I've set jumbo frames as per the manual.

Not really sure what else to do, I usually use USB cameras.

`$ ./tcam-gigetool info 15320011                                                                                                                                                         ✔ 

Model:    DFK 39GX265-Z20
Serial:   15320011
Firmware: Rev 2596, FPGA GigE3LZ-IMX:13
UserName: 

MAC Address:        00:07:48:42:39:2b
Current IP:         10.10.11.100
Current Subnet:     255.255.0.0
Current Gateway:    0.0.0.0

DHCP is:   disabled
Static is: enabled

Persistent IP:      10.10.11.100
Persistent Subnet:  255.255.0.0
Persistent Gateway: 0.0.0.0
`

TIS-Stefan, 2023-05-30:
Did you compile tiscamera with CMake setting `-DTCAM_BUILD_ARAVIS=ON`? If not, then the Aravis library is not used and the GigE camera wont be handled.

ShadeTechnik, 2023-05-30:
I don't recall whether I used any specific cmake options, I did it a few weeks ago. I did certainly check here before compiling.
[https://www.theimagingsource.com/en-us/documentation/tiscamera/building.html#configuring](url)

According to that Arivis is on by default. Is that not the case? Any way to check if Arivis support is enabled?

TIS-Stefan, 2023-05-30:
Hello
You can check for Aravis being used by:
```
$ tcam-ctrl --version
Versions: 
	Tcam:	1.1.0_master/70abad2a_rev_4139
	Aravis:	0.8_version_e977fa4
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc_tests
```
What is the IP address of your network controller? Does it match the camera's IP?

Stefan

ShadeTechnik, 2023-05-30:
```
$ ./tcam-ctrl --version                                                                                                                                                                 ✔ 
Versions: 
        Tcam:   1.1.1_master/ee45fcf7_rev_4142
        Aravis: 0.8_version_e977fa4
        Modules:        gst_aravis_arvu3v_v4l2_libusb_tools_doc
```
```
$ ip address 
3: enp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc mq state UP group default qlen 1000
    link/ether 00:e0:4c:04:b2:16 brd ff:ff:ff:ff:ff:ff
    inet 10.10.11.101/16 brd 10.10.255.255 scope global noprefixroute enp3s0
       valid_lft forever preferred_lft forever
    inet6 fe80::d247:afd5:506d:febc/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```
```
$ ./tcam-gigetool info 15320011                                                                                                                                                         ✔ 

Model:    DFK 39GX265-Z20
Serial:   15320011
Firmware: Rev 2596, FPGA GigE3LZ-IMX:13
UserName: 

MAC Address:        00:07:48:42:39:2b
Current IP:         10.10.11.100
Current Subnet:     255.255.0.0
Current Gateway:    0.0.0.0

DHCP is:   disabled
Static is: enabled

Persistent IP:      10.10.11.100
Persistent Subnet:  255.255.0.0
Persistent Gateway: 0.0.0.0

```

This is all I get launching tcam-capture. It does launch but shows no cameras.
`$ ./tcam-capture                                                                                                                                                                        ✔ 
2023-05-30T10:52:01 - unknown:0: warning libpng warning: iCCP: known incorrect sRGB profile
2023-05-30T10:52:04 - unknown:0: info No device selected
`

TIS-Stefan, 2023-05-30:
Hello 
I checked your network settings on my computer and it works perfectly. 
Thus, we must check a little bit deeper:
Please enable debug output:

```
export GST_DEBUG=tcam*:5
# if set output will not contain color codes
export GST_DEBUG_NO_COLOR=1
# log all output to this file
export  GST_DEBUG_FILE=/tmp/tcam-gst.log
# for separate log files an own handler has to be implemented
tcam-ctrl -l
``` 
(https://www.theimagingsource.com/en-us/documentation/tiscamera/environment.html?highlight=debug#gst-debug)
Instead of tcam-ctrl you can start tcam-capture.
If this gives not enough information, we may need to install wireshark and run tcam-ctrl -l or tcam-capture and log the network traffic.

Stefan

ShadeTechnik, 2023-05-30:
I managed to get tcam-capture working by recompiling and installing, I was just executing the binaries in the above examples. I don't know exactly why I didn't install the first time, I may have had errors that I fixed afterwards. After recompiling I was able to run make install without error and tcam-capture found the camera immediately.

I have a couple of questions about this camera now that I am able to use it. My application is for observing objects much closer than the lens is specified for (about 280mm) but I wanted to test this anyway as I need the motorized optic. I am able to focus to about 650/1000 but the focus is not consistent when zooming out. I've used a similar setup with Comutar lenses and if focused @ the closest zoom, the widest zoom will still be in focus without moving the focus.

 The motorized zoom on this seems to work really well and is obviously convenient to control, is it possible to get a modified focal length?

TIS-Stefan, 2023-06-07:
Hello

I am sorry for my US colleagues not answering. I saw your email from contact form coming in, but I do not know, whether they handled this, because we use different databases. However, I notified them.

Stefan

TIS-Stefan, 2023-06-22:
Hello

I spoke with my colleagues: Your camera must be sent in, so its front panel is replaced by a panel with a filter thread. 
On the other hand, there is not really anything to take care of in order to place a lens in front of the camera, because the zoom and focus can be controlled by software and thus adapted to the lens.

Stefan

ShadeTechnik, 2023-06-22:
I have this camera on evaluation so I'm not sure how that would work out. Maybe it's possible to send a modified camera for evaluation instead, under a new part number or something? Is it possible for you to reach out to the US teamand sort this out?

Since the camera is on evaluation, I'm not going to attempt to take it apart but it would certainly be alot easier if they could just send me the adapter face so I can replace it myself assuming the camera isn't assembled in some complicated way.

TIS-Stefan, 2023-06-23:
Hello
In this case, I suggest to try my US colleagues again.  I am located in Germany. 
Stefan

ShadeTechnik, 2023-06-23:
Stefan, I have been dancing around the issue but I think you are aware that there are no technical personnel in the US office, this was reiterated when I just tried to get someone on the phone there again just minutes ago. I have been trading emails with someone who is not terribly familiar with the technical aspects of these cameras. 

This has been the case since I purchased my first TIS camera in 2017. I do not mean to speak badly of TIS but continuously telling me I need to speak to someone in the US office when there is no one to speak to is getting me nowhere. If this person exists in the US they certainly will not get on the phone with me or respond via email. After mentioning all of the stuff from this discussion the response was:

 "In regards to modifying the camera or opening it up in any manner, this will void the eval. However, this additional lens will be external to the camera, so it will not require opening the camera up in any way. "

3 times I iterated that this camera has no external lens interface, this person is not understanding the issue and obviously is not looking at the camera they sent me. I would not be discussing this issue on a Github repo if there was someone else to talk to but you (Stefan) have literally been the only person at TIS to ever answer a single technical question I ever had. If there is a US person to speak to I would appreciate their direct contact info otherwise can you or someone else reach out to me at jwhite@shadetechnik.com because this is getting a bit frustrating now. Support tickets are not getting me anywhere either.

TIS-Stefan, 2023-06-26:
Hello Justin

I contacted them. However, They are in in US, while I am in Germany, so I am not their superior.

Stefan

ShadeTechnik, 2023-11-20:
Stefan,

Do you have any information on the camera I sent back for the face modification? It was supposed to take about 3 weeks, it's been about 4 months. Emailing the US people is not really getting me anywhere. I'm pretty far behind in testing this camera.

TIS-Stefan, 2023-11-21:
Hello

I have no information about this. Please contact me at https://www.theimagingsource.com/en-de/. Then I may can check, what happened.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar