TheImagingSource/tiscamera #524: MIPI camera not detected on raspberry pi compute module

yoanou, 2023-06-05:
I'm using a Raspberry Pi Compute Module 4 interfaced with the 4CM4-NANO-B board and I can't detect the camera connected with the MIPI interface.

On a fresh 32 bits Debian Bullseye, I installed: 
-  gstreamer1.0
-  tiscamer_1.1.04139_armhf_pi4_raspbian_10.deb
-  tcamdutils_1.0.0.560_armhf_pi4.deb
-  tcampimipisrc_1.3.0.662_armhf.deb

I then rebooted the raspberry pi.

But when i open tcam-capture, there is no device to open. And when I try to open the camera with gstreamer and tcampimipisrc ```gst-launch-1.0 tcampimipisrc! videoconvert ! ximagesink```, I have the following error :
```
Setting pipeline to PAUSED ...
ERROR: from element /GstPipeline:pipeline0/GstTcamPiMipiSrc:tcampimipisrc0: Failed to open camera.
Additional debug info:
/var/lib/jenkins/workspace/Linux/pi-mipi-cam/src/gst-pimipicam-src/tcampimipisrc_gst_element.cpp(474): impl_gst_open_camera (): /GstPipeline:pipeline0/GstTcamPiMipiSrc:tcampimipisrc0:
Underlying error value: Failed to find a device on the i2c bus
ERROR: pipeline doesn't want to preroll.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
```

I use :
 - OS: Raspbian, version 11 bullseye
 - Camera  model : DFM 36MX290-ML
 - tiscamera version : tcam : 1.1.0_master/70abad2a_rev_4139
 - I2C interface is enabled

To find out if the problem is with the MIPI connector, I tried to connect a V2.1 raspberry pi camera module. And it worked by adding ```dtoverlay=imx219,cam0``` in /boot/config.txt.

I also tried ```i2cdetect```, and here is the result :
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: 20 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --
```

Is there any way of getting a MIPI camera to work on a compute module?

TIS-Stefan, 2023-06-06:
Hello

We could see the debug output. Please do
```Shell
export GST_DEBUG=tcampimipisrc:5
```
and run the pipeline again and post the output.

Stefan

yoanou, 2023-06-06:
Here is the debug output:
```
Setting pipeline to PAUSED ...
0:00:00.103184248  1194  0x1012ea0 DEBUG          tcampimipisrc tcampimipisrc_gst_element.cpp:533:gst_virtual_func_change_state:<tcampimipisrc0> > State change to: GST_STATE_CHANGE_NULL_TO_READY
0:00:00.103261617  1194  0x1012ea0 INFO           tcampimipisrc tcampimipisrc_gst_element.cpp:408:impl_gst_open_camera_:<tcampimipisrc0> Opening device for serial=''.
0:00:00.104008252  1194  0x1012ea0 DEBUG          tcampimipisrc sensor_dispatch.cpp:113:create_sensor_itf: Failed to establish sensor/fpd-link connection via i2c bus, no device connected?
0:00:00.104085584  1194  0x1012ea0 WARN           tcampimipisrc tcampimipisrc_gst_element.cpp:474:impl_gst_open_camera:<tcampimipisrc0> error: Failed to open camera.
0:00:00.104139991  1194  0x1012ea0 WARN           tcampimipisrc tcampimipisrc_gst_element.cpp:474:impl_gst_open_camera:<tcampimipisrc0> error: Underlying error value: Failed to find a device on the i2c bus
ERROR: from element /GstPipeline:pipeline0/GstTcamPiMipiSrc:tcampimipisrc0: Failed to open camera.
Additional debug info:
/var/lib/jenkins/workspace/Linux/pi-mipi-cam/src/gst-pimipicam-src/tcampimipisrc_gst_element.cpp(474): impl_gst_open_camera (): /GstPipeline:pipeline0/GstTcamPiMipiSrc:tcampimipisrc0:
Underlying error value: Failed to find a device on the i2c bus
ERROR: pipeline doesn't want to preroll.
0:00:00.155577534  1194  0x1012ea0 DEBUG          tcampimipisrc tcampimipisrc_gst_element.cpp:533:gst_virtual_func_change_state:<tcampimipisrc0> > State change to: GST_STATE_CHANGE_NULL_TO_NULL
0:00:00.155687680  1194  0x1012ea0 DEBUG          tcampimipisrc tcampimipisrc_gst_element.cpp:617:gst_virtual_func_change_state:<tcampimipisrc0> < State change done.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
```

TIS-Stefan, 2023-06-13:
Hello 

I am very sorry for the delay of my answer. We figured out, that the PI4 B uses CM1 and your board CM0. That is similar to which you wrote already. 
Unfortunately the pimipisrc looks for CM1. The programmers have currently no resources to change the pimipisrc, 

Stefan

yoanou, 2023-06-13:
Hello,

Just to make sure I've understood correctly.

The MIPI tiscamera uses 4 data channels like the CAM1 interface. It is therefore impossible to use it with the CAM0 interface, which only uses 2 data channels.

Forcing the use of the CAM0 interface isn't enough for the camera to work properly. The only solution would be to find a Compute Module 4 IO Board with a MIPI connector connected to the CAM1 interface?

Yoan

TIS-Stefan, 2023-06-14:
Hell Yoan
no. The CM4 has two MIPI camera interfaces: CM 0 and CM 1. The Raspberry PI4 has only one MIPI camera interface and that seems to be CM 1. Therefore, our PIMIPISrc accesses CM 1 only.  But your board has the MIPI connector on CM 0. Thus, the camera is not detected, because we look on the wrong place. 
I checked this with the CM4 dev board, which has both MIPI camera connectors. 

I am sorry for this issue.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar