TheImagingSource/tiscamera #525: GIGE Camera not visible in Ubuntu WSL 2

gmpbigsun, 2023-06-14:
I'm trying to access my GIGE Cameras (2x DFK 33GX265) through Ubuntu 22 WSL. When running _tcam-gigetool list_ it outputs "No cameras found."

 - OS: Windows 10 using WSL 2 with Ubuntu 22
 - DFK 33GX265
 - Tcam:   1.1.1_master/ee45fcf7_rev_4142 
 - Aravis: 0.8_version_e977fa4                                                                                             
 - I'm able to see my cameras using GigECam IP Config Software on Windows and I'm able to receive images. 

I've build tiscamera using the following config:
`cmake -DTCAM_BUILD_ARAVIS=ON -DTCAM_BUILD_TOOLS=ON -DTCAM_BUILD_LIBUSB=OFF -DTCAM_BUILD_V4L2=OFF ..`

I was wondering if you knew a way to use the cameras in WSL. It would help me a lot since it would make development using those cameras way easier for me. If there's no known way to you to make it work, I was wondering if there's the possibility to directly access the cameras in my Python and C++ programs by passing an IP instead of serial number.

TIS-Stefan, 2023-06-14:
Hello

It does not work in an easy way. WSL2 uses an NAT in order to separate the virtual machine from the host. This means, the device-discover broadcasts are not sent to the camera, because the NAT does not forward them. 
As long as you have no native Linux computer, you may use Orcacles Virtual Box and install an Ubuntu in it. 

Stefan

gmpbigsun, 2023-06-15:
Hey, that makes sense, thank you
It appears that there are efforts by Microsoft to enable Bridging between WSL and the Host. 
Maybe in the future it'll work that way.

closed 2023-06-15 by gmpbigsun