TheImagingSource/tiscamera #526: Cannot get image from DFK33GX178

asuka-ms, 2023-06-23:
I am not good at English, so I apologize if I am difficult to understand.

Right now I have the DFK33GX178 connected to an ethernet cable and then to a PC via a usb-to-ethernet conversion cable.
I followed the [tutorial in the "tiscamera 1.0.0 documentation"](https://www.theimagingsource.com/en-us/documentation/tiscamera/tutorial.html) to get set up.
When I run "tcam-capture" (the application starts up), no image is sent from the camera.
I think the camera is not recognized because the list does not appear when I run "tcam-ctrl -l".
Can someone please help me?

execution environment
---------------------------------
camera：DFK33GX178
OS：Ubuntu 18.04
ROS：melodic
---------------------------------

TIS-Stefan, 2023-06-23:
Hello

If the camera is not recognized, does not deliver images or has another problem,
please check the following:

**Status LEDs do not indicate operation readiness**
Verify camera power and connection: If all LEDs on the camera are off then the
camera has no power. Please use an external power supply or PoE.
Green LED is off and the yellow LED flashes. If the camera is connected to a 10
Mbit or 100 Mbit network, it will cease to function. Please make sure that all devices
between the camera and the PC, as well as the PC network card support GigE
(1000mBit/s) speed.
The camera is displayed in the GigECamera Config tool or by "tcam-gigetool list", but not in other
programs such as IC Capture or IC Measure or tcam-capture. This happens if there is an IP address conflict.
If the camera has an invalid IP address, in most cases it can still be found by the
tools. In the first step, you assign a temporary IP address to the camera to reboot it.
Afterwards, the camera can be configured to DHCP or a new valid IP can be assigned.

source: https://s1-dl.theimagingsource.com/api/2.5/packages/publication/whitepaper-camera/wpgigecamerastartup/4df6c9cb-9fb5-50d0-8b42-82259fd2124b/wpgigecamerastartup_1.0.en_US.pdf

Stefan

asuka-ms, 2023-06-23:
Thanks for your reply.
I will try what you have told me to do.
I will keep you posted on any progress or problems.

closed 2023-06-23 by TIS-Stefan

asuka-ms, 2023-07-10:
Hello. I got a board that supports GigE and connected the tiscamera with an Ethernet cable (CAT.6). However, when I run "tcam-gigetool list" I get the following error message: the tiscamera is getting power and both the green and yellow LEDs are on. I am concerned that the yellow LED is not blinking...
I get the same error when I run "tcam-gigetool list" after running "sudo sysctl -w net.ipv4.conf.all.rp_filter=0".
----------------------------------------------
Your system has active rp_filter.
This can prevent misconfigured cameras from being detected.
To temporarily disable rp_filter execute:
        sudo sysctl -w net.ipv4.conf.all.rp_filter=0

No cameras found.
----------------------------------------------

asuka-ms, 2023-07-10:
Oops, I think I may have found the cause. Please wait a moment.