TheImagingSource/tiscamera #527: tiscamera example doesn't work

MaxexTian, 2023-07-07:
1. I use camera type "DMM 37AX296-ML"
When I used the example 6 from the website(https://www.theimagingsource.com/en-us/documentation/tiscamera/examples.html). I received the following error: 
Traceback (most recent call last):
  File "//test.py", line 6, in <module>
    gi.require_version("Gst", "1.0")
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 126, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Gst not available

2. And when I download the install package tiscamera, there is no examples in it. 

May I get the instructions to fix the code and run the example on my raspberrypi successfully?

**Screenshots**

**computer used (please complete the following information):**
 - OS: [raspbian]
 - Camera  model you use
    (DMM 37AX296-ML)
 - tiscamera version (tiscamera Linux 32 Bit Camera Library for ARMHF Architecture --version1.1.0.4139)

If your bug report has potentially sensitive information, please write us directly at support@theimagingsource.com .

MaxexTian, 2023-07-10:
After I install the gstreamer package the example runs. it has the image.
But the "trigger" Botton cannot work and when I press "enter" to trigger the image the error comes as "the property is not implemented"
May I have the solution of the problem?

TIS-Stefan, 2023-07-11:
I am very sorry, but your camera does not have trigger support. The documentation says:

[image]

Since your model still has the name "37AX..." I must assume, it is board revision 1.

When did you got this camera?

Stefan

MaxexTian, 2023-07-11:
I'm sorry for the typing mistake before. The type of my camera is DMM 36AX296-ML, as the form you sent to me, it should have the trigger mode. But it still reports error when I try to trigger  it in example 6.

MaxexTian

MaxexTian, 2023-07-11:
Hi
Actually I bought DMM 36AX296-ML on Jun 16th 2023. we purchased it because the website stated it has trigger capability. we were given a box with DMM 37AX296-ML. We were told that the DMM 36 and DMM 37 are the same camera and just difference in inventory vs website enumeration. If the DMM 37 is not capable for a trigger, then an RMA process needs to be started.  

[Estimate_55618_Redacted.pdf](https://github.com/TheImagingSource/tiscamera/files/12021834/Estimate_55618_Redacted.pdf)
[Gmail - Imaging Source 36 vs. 37_Redacted.pdf](https://github.com/TheImagingSource/tiscamera/files/12021835/Gmail.-.Imaging.Source.36.vs.37_Redacted.pdf)

TIS-Stefan, 2023-07-12:
Hello
This is a bad situation. Can you have a look on the board revision of the sensor board? If version 1, then there is no trigger and we have to make an RMA. 
I am very sorry for this
Stefan

MaxexTian, 2023-07-12:
Hi
Here are the pictures of the board version and the identification of the product.
If there is no trigger mode on this board, please contact me and make an RMA soon.
Thanks for your help!

MaxexTian

[image]
[image]
[image]

TIS-Stefan, 2023-07-13:
Hello

it is "R1.00", which means revision 1.0. Thus, please contact your camera dealer.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar