TheImagingSource/tiscamera #528: Can't build project

IldarGreat, 2023-07-11:
I'm following this [Building](https://github.com/TheImagingSource/tiscamera#building-tiscamera)
When I type make, I getting this error:

[ 99%] Built target tcam-gige-daemon
[100%] Building CXX object tools/tcam-gigetool/CMakeFiles/tcam-gigetool.dir/main.cpp.o
[100%] Linking CXX executable ../../bin/tcam-gigetool
[100%] Built target tcam-gigetool
[100%] Generate HTML Documentation with Sphinx
/usr/bin/sphinx-build:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
Traceback (most recent call last):
  File "/usr/bin/sphinx-build", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/ildar/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3328, in <module>
    def _initialize_master_working_set():
  File "/home/ildar/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3302, in _call_aside
    f(*args, **kwargs)
  File "/home/ildar/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3340, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/ildar/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 631, in _build_master
    ws.require(__requires__)
  File "/home/ildar/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 968, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/ildar/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 829, in resolve
    dist = self._resolve_dist(
  File "/home/ildar/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 870, in _resolve_dist
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'Sphinx==1.8.5' distribution was not found and is required by the application
make[2]: *** [doc/CMakeFiles/doc.dir/build.make:71: doc/CMakeFiles/doc] Error 1
make[1]: *** [CMakeFiles/Makefile2:2148: doc/CMakeFiles/doc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to build folder in project
2. Type on make
3. See error

**Screenshots**
[image]

**Computer used :**
 - OS: [Ubuntu 20.04, Debian Testing]
 - Architecture/Platform: [amd64], [Desktop]
 - Python 3.8.10

TIS-Stefan, 2023-07-12:
Hello

I searched for the error message "The 'Sphinx==1.8.5' distribution was not found and is required by the application" in the internet and found this:
https://stackoverflow.com/questions/57422485/pkg-resources-distributionnotfound-the-sphinx-1-8-5-distribution-was-not-fou
That describes the situation of a Python module being installed by apt-get and a second time by pip. That leads to issues. You may follow the answer from Stackoverflow and try again. It was deinstalling Sphinxs.

Stefan

IldarGreat, 2023-07-12:
Thanks, the command sudo apt-get install --reinstall python3-sphinx worked for me

closed 2023-07-12 by IldarGreat