TheImagingSource/tiscamera #529: Minimum exposure on DFK 37BUX290 is 4ms instead of 2us

stebett, 2023-07-20:
**Issue Description:**

**Current Behavior:**
I own a DFK 37BUX290 camera, and I have noticed a limitation regarding the exposure time settings. Even if I attempt to set the exposure time below 4ms (milliseconds), the camera consistently captures images with an exposure time of 4ms. This issue is preventing me from obtaining accurate images with exposure times lower than 4ms.

**Expected Behavior:**
The expected behavior is that the camera should allow exposure times to be set below 4ms accurately. When I set the exposure time to a value like 2us (microseconds), I would expect the camera to capture images with the specified exposure time of 2us, enabling me to capture fast-moving objects or scenes that require very short exposures.

**Steps to Reproduce:**
1. Connect the DFK 37BUX290 camera to a computer.
2. Use the camera's provided software or API to set the exposure time to a value below 4ms, such as 2us.
3. Trigger the camera to capture an image
4. Turn on an LED in front of the camera 3ms after the trigger has been sent (3ms after the end of the trigger, not the beginning)
5. Observe the resulting captured image to check if the powered LED is shown

**Additional Information:**
To further investigate this issue, I performed a test by turning on an LED 4ms after sending the trigger signal to the camera, with the intention of capturing the LED in the image with an exposure time of 2us. However, the captured image still showed the powered LED, indicating that the camera's exposure time was not accurate.

**Environment:**
- Camera Model: DFK 37BUX290
- Camera Firmware Version: lates
- Camera Software/Driver Version: latest
- Operating System: Windows 10
- Software/Code Used to Control the Camera: IC Capture - Image Acquisition

**Note:** This issue is hindering my ability to perform my tasks with the camera, and I would greatly appreciate any assistance in resolving this exposure time limitation. Thanks in advance.

stebett, 2023-07-20:
> The minimum exposure time is 20µs  

Yes, sorry I misread that
> Did you set the highest frame rate?  

Yes, I just tried to set minimum resolution (640x480) with 313.9028 fps and the issue is still there. Interestingly, the minimum exposure time is down to 1.5ms with this setting (the old setting was full hd with 140fps)
> Please let me know the serial number of the camera, so I can check firmware.

44120535

Thank you very much for the quick answer.

TIS-Stefan, 2023-07-20:
Hello

I am sorry for giving the wrong answer this morning.

1.  IMX290 is a rolling shutter sensor. Thus, the timespan between exposure of first and last line of the sensor is 1/framerate seconds.
2. In trigger mode the IMX290 runs in Global Reset Release mode. Thus, when the trigger occurs, the complete sensor is light sensitive. Each line exposes, until it is readout. That means the last line is exposed with 1/framerate seconds exposure time. As a result, the image becomes brighter to the bottom.

Now you know,why you can see the LED light under the conditions you wrote.

How to proceed depends on the task you want to perform.

1. The camera is suitable if, the scene is completely dark and when the trigger occurs, a flash light, our your LED turns on for 2µs.
2. The camera is the wrong one, if the scene is well lit and you want a short exposure time. Then you need a global shutter sensor, such as IMX273. [DFK 37BUX273](https://www.theimagingsource.com/en-us/product/industrial/37u/dfk37bux273/)

You may speak with our sales personal about the camera and may exchange your current one, in case you purchased it recently.

I will be out of office until Tuesday 25th.

Stefan

stebett, 2023-07-20:
Thank you for the information. My task is to film a very well lit scene with a very fast movement happening, so I need a short exposure.

Could we work around this problem by triggering with a pwm with a long duty cycle and setting the cameras to be triggered by the end of the pwm instead of the beginning, so that the camera is forced to take the image in a shorter time?

TIS-Stefan, 2023-07-25:
Hello

I am very sorry, but you can not get around this issue with a rolling shutter camera. The time interval between first and last line exposure is determined by the frame rate. The camera can not be forced to run faster than maximum frame rate. Thus, the camera is not suitable for your task.

Sorry for not having a better answer.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar