TheImagingSource/tiscamera #531: only one snapshot is happening during recorsing using DMK37BUX273

rangeetmanna, 2023-08-13:
Dear Stefan,

I was using this command in ubuntu termnal: gst-launch-1.0 tcambin name=tcam0 ! video/x-raw,format=GRAY8, width=640, height=480,framerate=30/1,binning=2X2 ! tee name=teemodule teemodule. ! queue max-size-time=0 max-size-bytes=0 max-size-buffers=0 ! videoconvert n-threads=4 ! ximagesink sync=false teemodule. ! queue name=save-queue max-size-time=0 max-size-bytes=0 max-size-buffers=0 ! videoconvert n-threads=4 ! queue ! x264enc ! mp4mux ! queue max-size-time=0  max-size-bytes=0 max-size-buffers=0 ! filesink async=true sync=false name=save-sink location =/home/kevin/test.mp4 -e

,........Instead of recording a video, only one snapshot happened every time. It would be very helpful if you could suggest a solution. 

Thanking you!

Kind Regards

Rangeet

TIS-Stefan, 2023-08-14:
Hello
I would start with a more simple pipeline first and see, what happens:

gst-launch-1.0 tcambin name=tcam0 ! video/x-raw,format=GRAY8, width=640, height=480,framerate=30/1,binning=2X2 !  videoconvert ! x264enc ! mp4mux ! filesink async=true sync=false name=save-sink location =/home/kevin/test.mp4 -e

And see, what happens. 
I see, you show a live video window too. Do you see only one image there too?

Stefan

rangeetmanna, 2023-08-14:
Hi Stefan
I tried the new one, the code is running but no running window pops up. Its shows up to redistribute latency... in the terminal only.
Need help!

Thanks 

Rangeet

TIS-Stefan, 2023-08-15:
Hi Rangeet

there is no window, because the pipeline does not have one. More interresting: What is in the test.mp4 in kevins's directory?

Stefan

rangeetmanna, 2023-08-15:
Dear Stefan,
the video has been saved and it works fine. But, is it possible anyhow to record and view the video simultaneously?
That would be great!

Thanks!

Rangeet

TIS-Stefan, 2023-08-15:
Hello Rangeet

If video saving works fine now, start enhancing your GStreamer pipeline with an ximagesink as you tried at beginning. 
I suggest to to that step by step, so start with showing the video in the ximagesink, but do not add video saving. 
Then start adding the tee and so one. 
Do that step by step and always check the results you get. 
It is all GStreamer usage. It is not camera related.

Stefan

rangeetmanna, 2023-08-16:
Dear Stefan
With this code I can view the video in a window properly --------gst-launch-1.0 tcambin name=tcam0 ! video/x-raw,format=GRAY8, width=640, height=480,framerate=30/1,binning=2X2 ! videoconvert ! x264enc ! mp4mux ! filesink async=true sync=false name=save-sink location =/home/kevin/test.mp4 -e

With this code I can record a video properly------------gst-launch-1.0 tcambin name=tcam0 ! video/x-raw,format=GRAY8, width=640, height=480,framerate=30/1,binning=2X2 ! videoconvert ! x264enc ! mp4mux ! filesink async=true sync=false name=save-sink location =/home/kevin/test.mp4 -e

But, I could not able to combine both. Somehow error pops up or one shap shot happens. Need some solution ! 

Thanks !

Regards

Rangeet

TIS-Stefan, 2023-08-16:
Hello Rangeet

You need a "tee" filter : https://gstreamer.freedesktop.org/documentation/coreelements/tee.html?gi-language=c 
It splits the GStreamer Pipeline into branches. One for display and another for video capture. You used that already in your first post. 
Searching in the internet for "Gstreamer tee examples" you will find
https://stackoverflow.com/questions/45873407/gstreamer-tee-with-different-capabilities-on-each-branch
and more. 

It is GStreamer usage, not a camera issue.

Stefan

TIS-Stefan, 2023-08-18:
Hello Rangeet,
I am sorry, but I am not familiar with the queue parameters. You may look at https://gstreamer.freedesktop.org/documentation/coreelements/queue.html?gi-language=c for the documentation. 

Stefan

TIS-Stefan, 2023-08-18:
Hello Rangeet

I made some tests on my own. The pipeline is stripped down to the necessary parts and instead of the camera I use a videotestsrc:

```
gst-launch-1.0 videotestsrc name=tcam0 ! video/x-raw,format=BGRx, width=640, height=480,framerate=30/1 ! tee name=teemodule teemodule. ! queue ! videoconvert ! ximagesink sync=false teemodule. ! queue ! videoconvert ! x264enc ! mp4mux ! filesink sync=false name=save-sink location =./test.mp4
```
Thus: It is a pure GStreamer issue. 
So, lets see, what happens if we use jpegenc and avimux instead of x264enc:
```
gst-launch-1.0 videotestsrc name=tcam0 ! video/x-raw,format=BGRx, width=640, height=480,framerate=30/1 ! tee name=teemodule ! queue ! videoconvert ! xvimagesink sync=false teemodule. ! queue ! videoconvert ! jpegenc ! avimux ! filesink sync=false name=save-sink location =./test.avi
```
And.... it works perfectly. 
I am sorry, I have no solution for this issue. Thus, I suggest you ask the question with the videotestsrc pipeline above at https://stackoverflow.com/ or at https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues.

Stefan

rangeetmanna, 2023-08-22:
gst-launch-1.0 tcambin name=tcam0 ! video/x-raw,format=BGRx, width=640, height=480,framerate=30/1,binning=2X2 ! tee name=teemodule teemodule. ! queue max-size-time=0 max-size-bytes=0 max-size-buffers=0 ! videoconvert n-threads=4 ! ximagesink sync=false teemodule. ! queue name=save-queue max-size-time=0 max-size-bytes=0 max-size-buffers=0 ! videoconvert n-threads=4 ! queue ! jpegenc ! avimux ! queue max-size-time=0  max-size-bytes=0 max-size-buffers=0 ! filesink async=true sync=false name=save-sink location =/home/kevin/test.avi -e

TIS-Stefan, 2023-08-22:
Ah, ok, the JPEG AVI solution.

closed 2023-08-22 by TIS-Stefan