TheImagingSource/tiscamera #532: DZK 33GX250 not retaining IP configuration if gateway is set to 0.0.0.0

daykin, 2023-08-31:
**Describe the bug**
When configuring the IPv4 settings on a DZK 33GX250 Polarsens camera via GigEVision/aravis,with a gateway IP of 0.0.0.0, the IP configuration is not retained. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a DZK 33GX250, via PoE, to host machine. The camera appears with a link-local 169.254.xxx.xxx address.
3. Issue `tcam-gigetool rescue --ip=<desired IP on same subnet> --netmask 255.255.255.0 --gateway 0.0.0.0 <serial number>
4. Confirm `tcam-gigetool list` now shows the correct settings.
5. Issue `tcam-gigetool set --mode static <serial number>` [1][2]
6. Issue `tcam-gigetool set --ip <same IP address> --netmask 255.255.255.0 --gateway 0.0.0.0 <serial number>`
7. Power-cycle the camera by unplugging and re-plugging the PoE cable
8. Issue `tcam-gigetool list` once more, and the camera has returned to the link-local 169.254.xxx.xxx address.

**Expected behavior**
IP configuration persists in EEPROM across reboots. Since we want to dump any packets without a route, i.e. those not destined for the host server, we usually just set the gateway to 0.0.0.0. 

**Screenshots**
[image]

**computer used (please complete the following information):**
 - OS: Debian 12
 - Architecture/Platform: amd64 Desktop
 - Camera  DZK 33GX250
 - tiscamera version 
 - ```Versions: 
        Tcam:   1.1.1_uc5/de8b602_rev_39
        Aravis: unknown_version_external <- Editor's note: this is libaravis-dev 0.8.26-1 from Debian official repo
        Modules:        gst_aravis_arvu3v_v4l2_libusb_tools
**Additional context**
Issue does not occur if a nonzero gateway IP is set, so this is not necessarily a huge issue. However, it should be documented somewhere as I tried for ages to get the configuration to 'stick' without knowing this was the issue.
[1] If --mode is set with the rest of the IPv4 parameters, it errors with
```Too many arguments as potential camera identifier.
Arguments are: --mode static 46220267 ```
[2] Issue occurs regardless of ordering the ip/netmask/gateway and mode.

TIS-Stefan, 2023-09-01:
Hello
I ran
```
tcam-gigetool set --ip 192.168.1.113 --netmask 255.255.255.0 --gateway 0.0.0.0 --mode static 34810033
```
and the ip address was changed from DHCP to static as specified. After power cycling the camera it has the wanted IP addresses.  I wonder, why you got the argument error. 
If the --mode static is not sent with the IP address parameters, then nothing will happen. 

Stefan

daykin, 2023-09-01:
After recompiling and reinstalling everything, the "too many arguments" error is gone. But, the volatile 0.0.0.0 issue persists specifically for the 33GX250. Perhaps it's a firmware issue rather than a tiscamera issue?

closed 2023-09-01 by daykin