TheImagingSource/tiscamera #533: Black dashed lines across image

daykin, 2023-09-01:
**Describe the bug**
intermittent black 'dashed' lines appear in acquisition

**To Reproduce**
Begin acquisition on camera with tcam-capture, save image.

**Expected behavior**
Image appears without artifacts. 

**Screenshots**
[image]
oblem.

**computer used (please complete the following information):**
 - OS: Debian 12 
 - Architecture/Platform: amd64 Desktop
 - DMK 33GX174 (occurs with any camera)
 - tiscamera version tcam-ctrl --version
    Versions: 
            Tcam:   1.1.1_/_rev_
            Aravis: unknown_version_external <- libaravis-dev 0.8 from official debian repo
            Modules:        gst_aravis_arvu3v_v4l2_libusb_tools

**Additional context**
I am not sure which layer this problem originates in, perhaps it's an Aravis problem.

TIS-Stefan, 2023-09-01:
Hello
We are not sure, what causes these artifacts. But you could try to use a different MTU on the network controller and see, whether the spacing of the dots changes. This may gives a hint for the cause.
I am sorry for guessing. 
Stefan

daykin, 2023-09-01:
Hi Stefan, 
Thanks for your reply. It only occurs on one specific machine, so you might be correct that it has to do with the physical layer. I will check and report back.

daykin, 2023-09-01:
Yes, that fixed it. Dropping MTU to 1024 and returning it to 1500 made the lines disappear. Thanks for pointing this out!

closed 2023-09-01 by daykin

TIS-Stefan, 2023-09-04:
Hello
What happens, if you set the MTU to 9000? Is that at least possible on your computer? It will reduce network traffic.
Stefan

daykin, 2023-09-11:
Hi Stefan,

Bumping up to 9000 works just fine, but I am not sure all of our production network infrastructure supports jumbo frames or not. I'd have to check with our IT guy.