TheImagingSource/tiscamera #534: Can't set acquire period longer than 0.5s when capturing less than full region

daykin, 2023-09-11:
I guess this is not strictly a bug, per se. But, on our DZK 33GX250, the acquire period cannot be set longer than 0.5s in some cases it's 0.5s max, but with a larger region size, I can set it up to 1.0s. The [Documentation](https://s1-dl.theimagingsource.com/api/2.5/packages/documentation/manual-trm/trmdzk33gx250/e85ec63e-bbd3-53e6-82e3-7b72baf150d1/trmdzk33gx250.en_US.pdf) pp 16-17 Indicates some minimum depending on pixel format, sensor size and readout mode,  and there's a corresponding table for maximum, but not minimum. Is there a table with this info?

TIS-Kevin, 2023-09-21:
Dear Daykin,
sorry for my late reply.
There is no Information about the slowest Framerate like a Table or something like this.
The Framerates are also like that in other GigE Cameras.

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar