TheImagingSource/tiscamera #535: Cannot get image/video from DFK 42BUC03 on Linux

FewerToWhom, 2023-10-20:
**Describe the bug**
Even after installing the drivers from the .deb packaged, I cannot get any pictures on Linux (Ubuntu 22.04).
The camera works on Windows no problem.

The device shows up on lsusb:
```
$ lsusb 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 06cb:00f0 Synaptics, Inc. 
Bus 003 Device 002: ID 04f2:b6bf Chicony Electronics Co., Ltd HP HD Camera
Bus 003 Device 005: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 012: ID 199e:8208 The Imaging Source Europe GmbH DFK 42BUC03
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

but not at all as /dev/video. There I only see the builtin laptop camera. `tcam-ctrl -l` gives me nothing.

After going through other issues posted here, I've changed the camera from running on proprietary mode to uvc via the
firmware update tool:
```
$ ./tcam-firmware-update -id 48410323

Device manufacturer: The Imaging Source Europe GmbH
Product name:        DFK 42BUC03
Serial number:       48410323
VendorID:ProductID:  199e:8308
Firmware version:    135
UVC mode is:         on
Camera EEPROM size:  32768`

Then it shows up on /dev/video, and also on `tcam-ctrl -l`. However, `tcam-capture` gives me only a black screen.

In case this is of any use, here is the output of v4l2 (video0 to video3 correspond to the builtin cameras):
`$ v4l2-ctl -d /dev/video4 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : DFK 42BUC03
	Bus info         : usb-0000:00:14.0-1
	Driver version   : 6.2.16
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : DFK 42BUC03
	Serial           : 48410323
	Bus info         : usb-0000:00:14.0-1
	Media version    : 6.2.16
	Hardware revision: 0x00000813 (2067)
	Driver version   : 6.2.16
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : DFK 42BUC03
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000010: from remote pad 0x100000a of entity 'Extension 4' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Input 1: ok)
Format Video Capture:
	Width/Height      : 1280/720
	Pixel Format      : 'GRBG' (8-bit Bayer GRGR/BGBG)
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 921600
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 720
	Default     : Left 0, Top 0, Width 1280, Height 720
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0

User Controls

                           gain 0x00980913 (int)    : min=34 max=255 step=1 default=16 value=36

Camera Controls

         exposure_time_absolute 0x009a0902 (int)    : min=1 max=300000 step=1 default=127 value=127
                        privacy 0x009a0910 (bool)   : default=0 value=0
                        trigger 0x0199e208 (bool)   : default=0 value=0
                        binning 0x0199e925 (int)    : min=1 max=1 step=1 default=1 value=1
                       x_offset 0x0199e927 (int)    : min=0 max=64224 step=2 default=0 value=656
                       y_offset 0x0199e928 (int)    : min=0 max=64552 step=2 default=0 value=612
   trigger_global_reset_shutter 0x0199e92a (bool)   : default=0 value=0
```

I have also tried the following command
```
$ v4l2-ctl -d /dev/video4 --set-fmt-video=width=1280,height=720,pixelformat=GRBG --stream-mmap --stream-count=1 --stream-to=FILE.raw
```
It works fine for the builtin cameras, but not for the DFK one.

Trying gst-launch itself with (some?) debug information gave me
```
$ TCAM_LOG=DEBUG gst-launch-1.0 --gst-debug=tcamsrc:5,tcambin:5 --gst-debug-no-color tcambin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
0:00:00.015384432 12124 0x55d032b27470 INFO                 tcambin gsttcambin.cpp:384:select_transform_element: Auto selected transform element: tcamdutils
0:00:00.020812822 12124 0x55d032b27470 INFO                 tcamsrc gsttcamsrc.cpp:340:open_source_element:<tcambin-source> Opened device with serial: '48410323' type: 'v4l2'
0:00:00.020838605 12124 0x55d032b27470 INFO                 tcambin gsttcambin.cpp:183:tcambin_create_source:<tcambin0> Opened device has serial: '48410323' type: 'v4l2'
0:00:00.020858148 12124 0x55d032b27470 INFO                 tcambin gsttcambin.cpp:435:tcambin_create_elements:<tcambin0> creating all tcambin pipeline elements
0:00:00.021215663 12124 0x55d032b27470 DEBUG                tcambin gsttcambin.cpp:399:create_and_add_element:<tcambin0> Adding tcamdutils(0x55d032b6a0d0) to pipeline
0:00:00.022752386 12124 0x55d032b27470 DEBUG                tcambin gsttcambin.cpp:554:tcambin_create_elements:<tcambin0> Internal pipeline: tcamsrc ! capsfilter ! tcamdutils
0:00:00.022878620 12124 0x55d032b27470 INFO                 tcambin gsttcambin.cpp:659:gst_tcam_bin_change_state:<tcambin0> caps of sink: video/x-raw, format=(string)YV12, width=(int)[ 1, 16384 ], height=(int)[ 1, 16384 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)I420, width=(int)[ 1, 16384 ], height=(int)[ 1, 16384 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)NV12, width=(int)[ 1, 16384 ], height=(int)[ 1, 16384 ], framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, width=(int)[ 1, 16384 ], height=(int)[ 1, 16384 ], framerate=(fraction)[ 0/1, 2147483647/1 ], format=(string){ ABGR64_LE, BGRA64_LE, AYUV64, ARGB64_LE, ARGB64, RGBA64_LE, ABGR64_BE, BGRA64_BE, ARGB64_BE, RGBA64_BE, GBRA_12LE, GBRA_12BE, Y412_LE, Y412_BE, A444_10LE, GBRA_10LE, A444_10BE, GBRA_10BE, A422_10LE, A422_10BE, A420_10LE, A420_10BE, RGB10A2_LE, BGR10A2_LE, Y410, GBRA, ABGR, VUYA, BGRA, AYUV, ARGB, RGBA, A420, AV12, Y444_16LE, Y444_16BE, v216, P016_LE, P016_BE, Y444_12LE, GBR_12LE, Y444_12BE, GBR_12BE, I422_12LE, I422_12BE, Y212_LE, Y212_BE, I420_12LE, I420_12BE, P012_LE, P012_BE, Y444_10LE, GBR_10LE, Y444_10BE, GBR_10BE, r210, I422_10LE, I422_10BE, NV16_10LE32, Y210, v210, UYVP, I420_10LE, I420_10BE, P010_10LE, NV12_10LE32, NV12_10LE40, P010_10BE, Y444, RGBP, GBR, BGRP, NV24, xBGR, BGRx, xRGB, RGBx, BGR, IYU2, v308, RGB, Y42B, NV61, NV16, VYUY, UYVY, YVYU, YUY2, I420, YV12, NV21, NV12, NV12_64Z32, NV12_4L4, NV12_32L32, Y41B, IYU1, YVU9, YUV9, RGB16, BGR16, RGB15, BGR15, RGB8P, GRAY16_LE, GRAY16_BE, GRAY10_LE32, GRAY8 }
Pipeline is live and does not need PREROLL ...
INFO:
Working with src caps: video/x-bayer, framerate=(fraction)15/2, width=(int)2592, height=(int)1944, binning=(string)1x1, skipping=(string)1x1, format=(string)grbg
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstXvImageSink:autovideosink0-actual-sink-xvimage: Output window was closed
Additional debug info:
../sys/xvimage/xvimagesink.c(568): gst_xv_image_sink_handle_xevents (): /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstXvImageSink:autovideosink0-actual-sink-xvimage
Execution ended after 0:00:44.047892677
Setting pipeline to NULL ...
Freeing pipeline ...
```

but again a black screen...
Additionally, and I don't know if this is relevant, in uvc mode the camera produces a black image also on Windows.

I have also tried building tiscamera from source, but that fared no better than using the .deb installation.

**computer used:**
 - OS: Ubuntu 22.04
 - Architecture/Platform: amd64
 - Camera  DFK 42BUC03
 - tiscamera version:
`Versions: 
	Tcam:	1.1.1_master/0be2537f_rev_4143
	Aravis:	0.8_version_e977fa4
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc`

TIS-Stefan, 2023-10-20:
Hello

The camera needs a firmware update in order to become v4l2 compatible.
Please clone the repository from https://github.com/TheImagingSource/tcam-firmware-update
The firmware to be written into your camera is /firmware/usb2/dfk42uc03_3014.euvc.

FewerToWhom, 2023-10-20:
Many thanks! If only I had had the guts to try updating the firmware sooner... :D

closed 2023-10-20 by FewerToWhom