TheImagingSource/tiscamera #537: Broken dependency

leonhartyao, 2023-10-31:
**Describe the bug**
Error while running the script "dependency-manager" to install dependency. Some dependency specification might be out of date.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to repo folder
2. `sudo ./scripts/dependency-manager install`
4. See error

```bash
Executing: sudo apt-get install git g++ cmake pkg-config libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev libgirepository1.0-dev libusb-1.0-0-dev libzip-dev libzip5 libglib2.0-0 libgirepository-1.0-1 libusb-1.0-0 libuuid1 uuid-dev libudev-dev libudev1 python3-sphinx libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly qtgstreamer-plugins-qt5 libxml2-dev libxml2 ninja-build libqt5widgets5 libqt5gui5 qtbase5-dev graphviz

Reading package lists... Done
Building dependency tree       
Reading state information... Done
g++ is already the newest version (4:9.3.0-1ubuntu2).
g++ set to manually installed.
libusb-1.0-0 is already the newest version (2:1.0.23-2build1).
libusb-1.0-0-dev is already the newest version (2:1.0.23-2build1).
libusb-1.0-0-dev set to manually installed.
pkg-config is already the newest version (0.29.1-0ubuntu4).
graphviz is already the newest version (2.42.2-3build2).
gstreamer1.0-plugins-ugly is already the newest version (1.16.2-2build1).
gstreamer1.0-plugins-ugly set to manually installed.
libzip-dev is already the newest version (1.5.1-0ubuntu1).
libzip-dev set to manually installed.
libzip5 is already the newest version (1.5.1-0ubuntu1).
libzip5 set to manually installed.
qtgstreamer-plugins-qt5 is already the newest version (1.2.0-5).
qtgstreamer-plugins-qt5 set to manually installed.
git is already the newest version (1:2.25.1-1ubuntu3.11).
gstreamer1.0-plugins-base is already the newest version (1.16.3-0ubuntu1.2).
gstreamer1.0-plugins-good is already the newest version (1.16.3-0ubuntu1.2).
libglib2.0-0 is already the newest version (2.64.6-1~ubuntu20.04.6).
libglib2.0-dev is already the newest version (2.64.6-1~ubuntu20.04.6).
libglib2.0-dev set to manually installed.
libgstreamer1.0-0 is already the newest version (1.16.3-0ubuntu1.1).
libxml2 is already the newest version (2.9.10+dfsg-5ubuntu0.20.04.6).
libxml2-dev is already the newest version (2.9.10+dfsg-5ubuntu0.20.04.6).
libxml2-dev set to manually installed.
libqt5gui5 is already the newest version (5.12.8+dfsg-0ubuntu2.1).
libqt5widgets5 is already the newest version (5.12.8+dfsg-0ubuntu2.1).
qtbase5-dev is already the newest version (5.12.8+dfsg-0ubuntu2.1).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
libgirepository-1.0-1 is already the newest version (1.64.1-1~ubuntu20.04.1).
libudev-dev is already the newest version (245.4-4ubuntu3.22).
libudev-dev set to manually installed.
libudev1 is already the newest version (245.4-4ubuntu3.22).
libuuid1 is already the newest version (2.34-0.1ubuntu9.4).
uuid-dev is already the newest version (2.34-0.1ubuntu9.4).
uuid-dev set to manually installed.
gstreamer1.0-plugins-bad is already the newest version (1.16.3-0ubuntu1).
gstreamer1.0-plugins-bad set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgstreamer-plugins-base1.0-dev : Depends: libx11-xcb-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

**Expected behavior**
All packages are installed successfully.

**computer used (please complete the following information):**
 - OS: [Ubuntu 20.04]
 - Architecture/Platform: [amd64], [Desktop]

TIS-Kevin, 2023-11-01:
Dear leonhartyao,

did you already try to install libx11-xcb-dev manually? Which Error Code is shown there?
Also does it help to do sudo apt update and try to install again?

Kevin

leonhartyao, 2023-11-06:
Dear Kevin,

Thanks for your quick reply. I have updated the cache but `sudo apt install libx11-xcb-dev` gives following error:

```bash
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libx11-xcb-dev : Depends: libx11-xcb1 (= 2:1.6.9-2ubuntu1.6) but 2:1.7.2-11~ubuntu20.04 is to be installed
E: Unable to correct problems, you have held broken packages.
```

leonhartyao, 2023-11-06:
My workaround is installing libx11-xcb-dev with `sudo aptitude install libx11-xcb-dev` and downgrading the version.
This fixed my issue but still a temporary solution.

TIS-Stefan, 2023-11-06:
Hi
I found the same solution at https://askubuntu.com/questions/1275725/while-installing-libx11-dev-e-unable-to-correct-problems-you-have-held-broken.
It rather is an issue of Ubuntu than of tiscamera. Therefore, I close this issue. Re-open it, if you have more questions.

Stefan

closed 2023-11-06 by TIS-Stefan