TheImagingSource/tiscamera #538: Failed to find any device to open DMK23G445

leonhartyao, 2023-11-06:
**Describe the bug**
I was trying to verify the installation for two DMK23G445 cameras. They are responsive to `tcam-gigetool`. However, there is no device listed in `tcam-capture`. 

- Running `gst-launch-1.0 tcambin ! videoconvert ! ximagesink` returns the following error:

```bash
gst-launch-1.0 --gst-debug=tcamsrc:5 tcambin ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
0:00:00.033249189 52281 0x555ae5c6ec60 WARN                 tcamsrc gsttcamsrc.cpp:249:open_source_element:<tcambin-source> error: Failed to find any device to open.
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Failed to find any device to open.
Additional debug info:
/home/yac1nst/repos/tiscamera/src/gstreamer-1.0/tcamsrc/gsttcamsrc.cpp(249): open_source_element (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source
Setting pipeline to NULL ...
Freeing pipeline ...
```

- `tcam-ctrl -l` returns nothing
- ping successful for both cam
- `tcam-gigetool list` shows both camera with correct info
```bash
tcam-gigetool list
Model Name           | Serial Number    | User Defined Name    | Current IP       
DMK 23G445             216106**           A2                     192.168.1.4        
DMK 23G445             216106**           A1                     192.168.1.59
```

- `tcam-gigetool check-control` said that both cameras are not controlled by anyone.

**computer used (please complete the following information):**
 - OS: Ubuntu 20.04
 - Architecture/Platform: amd64, Desktop
 - Camera  model you use
 - tiscamera version (tcam-ctrl --version, if possible)
        Tcam:	1.1.1_master/0be2537f_rev_4143
	Aravis:	0.8_version_e977fa4
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc

leonhartyao, 2023-11-06:
Just saw that the current version doesn't support the cam model. I have to use older version.

closed 2023-11-06 by leonhartyao

TIS-Stefan, 2023-11-06:
The current version can support your camera too:
```
export TCAM_DISABLE_DEVICE_BLACKLIST=1
```
(Source: https://www.theimagingsource.com/en-us/documentation/tiscamera/environment.html?highlight=environment#tcam-disable-device-blacklist)

Stefan