TheImagingSource/tiscamera #542: Is there a DFK31BU camera driver available for the Raspberry Pi?

Noz0mizore, 2024-01-30:
I'm trying to connect a camera to a Raspberry Pi 4 for a project, but I don't see any compatible drivers. Is there any way to connect this camera to the Raspberry Pi in some way and get the trigger to work?

As for the OS version, it doesn't matter what version you use as long as you can get the camera to work.

TIS-Stefan, 2024-01-30:
There is no driver needed, but the camera does not identify itself as UVC/V4l2 device. I made a manual including the rules file at
[The Imaging Source Knowledgebase](https://wiki.theimagingsource.com/index.php/Using_USB_2.0_CCD_Cameras_in_Linux)

Noz0mizore, 2024-01-31:
> There is no driver needed, but the camera does not identify itself as UVC/V4l2 device. I made a manual including the rules file at [The Imaging Source Knowledgebase](https://wiki.theimagingsource.com/index.php/Using_USB_2.0_CCD_Cameras_in_Linux)

So how do I use the camera's external trigger?
Oh, and when I browse the device with the lsusb command, it outputs the model name fine, but it doesn't actually work?

Noz0mizore, 2024-02-05:
``tcam-capture
2024-02-05T11:30:01 - unknown:0: warning libpng warning: iCCP: known incorrect sRGB profile
2024-02-05T11:30:01 - unknown:0: warning QXcbConnection: XCB error: 148 (Unknown), sequence: 191, resource id: 0, major code: 140 (Unknown), minor code: 20
2024-02-05T11:30:26 - unknown:0: info device selected: --``

When I run tcam, I get this error
ls /dev/video0 is recognized normally

TIS-Stefan, 2024-02-05:
Hello
The quoted messages are warnings. 
Are you sure, you have a 31BU and not a 21BU with 640x480 resolution only? Can you please give me the serial number of your camera, so I can check?
Does tcam-capture show a live video?

Noz0mizore, 2024-02-07:
[image]