TheImagingSource/tiscamera #543: Intermittent failure to change pipeline state

teawizardry, 2024-02-04:
I'm using a DMM 37UX250-ML camera board with a Raspberry Pi 4. Usually, it has no issues starting the pipeline, but after a repeated number of times of changing states (Playing to Paused to Ready and back again) I get the following error:
```
tcambin name=source ! capsfilter name=caps !  appsink name=sink
video/x-raw,format=GRAY8,width=1216,height=1024,binning=2x2,framerate=60/1
Error starting pipeline. (<enum GST_STATE_CHANGE_ASYNC of type Gst.StateChangeReturn>, state=<enum GST_STATE_PAUSED of type Gst.State>, pending=<enum GST_STATE_PLAYING of type Gst.State>)
```
I never have any errors going from Ready to Paused. What could be causing it to hang? 

I'm trying to catch the behavior in a log file, but I think I didn't have the debug level set high enough. I'll update with that when I get it.

TIS-Stefan, 2024-02-05:
Hello

Can you please let me know the tcambin version? Run tcam-ctrl -l
Also can you please post a minimal reproducible sample script? Even if I have an idea about what you are doing? 

Thank you in advance!
Stefan

teawizardry, 2024-02-05:
Sure!

tcambin version:
```
Versions: 
        Tcam:   1.1.1_master/ee45fcf7_rev_4142
        Aravis: 0.8_version_e977fa4
        Modules:        gst_aravis_arvu3v_v4l2_libusb_tools_doc_tests
```
I am not using aravis.

Camera initialization:
```
# initialize callback function and TIS
IC = ImageCallback(False, None)
Tis = TIS()

# Set callback function and open device.
Tis.set_image_callback(on_new_image, self.IC)

# select camera 
Tis.open_device(serial_number, height, width, framerate, SinkFormats.GRAY8, binning=binning, showvideo=False)

# Start Pipeline
Tis.set_property("TriggerMode", "Off")
IC.busy = True
Tis.start_pipeline()

# Set Properties
Tis.set_property("TriggerMode", "On")
# and set whatever other properties here

time.sleep(0.1) 
IC.busy = False
````
I slightly modified the Gst caps function to enable binning:
```
caps = Gst.Caps.from_string('video/x-raw,format=%s,width=%d,height=%d,binning=2x2,framerate=%s' % (self.sinkformat.value, self.width, self.height, self.framerate))
```

When I take an picture right now I'm simply using software triggers:
```
Tis.execute_command("TriggerSoftware")
```

And then for starting and stopping the camera I go from Ready => Paused => Playing then Playing => Paused => Ready.
```
def start_pipeline(self):
    """
    Start the pipeline, so the video runs
    """
    self._setcaps()
    if self.pipeline.get_state(5000000000)[1] != Gst.State.READY:
        self.pipeline.set_state(Gst.State.READY)
    self.pipeline.set_state(Gst.State.PAUSED)
    self.pipeline.set_state(Gst.State.PLAYING)
    error = self.pipeline.get_state(5000000000)
    if error[1] != Gst.State.PLAYING:
        print("Error starting pipeline. {0}".format(error))
        return False
    return True
    
def stop_pipeline(self):
    self.pipeline.set_state(Gst.State.PAUSED)
    self.pipeline.set_state(Gst.State.READY)
```

So in total my main loop is like:
```
# start camera
cam = Camera(serial_number = "12345",
             height = 1216,
             width = 1024,
             framerate = "60/1",
             binning = True)

# take pictures
foo()

# on exit
Tis.stop_pipeline()
```

The hang up happens both when stopping and starting the camera in the main code or when running the program multiple times. Right now I have eliminated starting and stopping the camera in the same program to try to isolate the error.

TIS-Kevin, 2024-02-07:
Hello teawizardry,

i tried to reproduce your issue but it is working on my machine.
But for some code optimization you can skip the step of state paused. Because if you set from playing to ready it will walk also through paused.

As you mentioned before "I'm trying to catch the behavior in a log file, but I think I didn't have the debug level set high enough. I'll update with that when I get it."

for logging you should use GST_DEBUG=*:3,tcam*:5 
If you could let us know what the logging says we appreciate it.

Thank you in advance!
Kevin

TIS-Kevin, 2024-02-07:
Additionally i like to add something to my last message.
Because i saw you are working on PI with a usb camera. Is the camera connected directly to the Pi or is it connected to a powered usb hub?
In case of the direct connection there may be some issues with the power supply for the camera. That could cause the problem you occure.

teawizardry, 2024-02-07:
It is connected directly to the Pi. Is there a way to see if power supply issues are the problem? And if so, how much power does it need?

TIS-Kevin, 2024-02-08:
Hello teawizardry,
well there can be issues caused by the power supply to the Pi but typically are power saving mechanisms.
You may check this Website which provides already many informations about power supply problems with a Pi:
https://pimylifeup.com/raspberry-pi-low-voltage-warning/

However, as i can see you are using tiscamera 1.1 and the udev rules should disable the power saving mechanisms for our cameras.

To be on the sure site i suggest to connect the camera to a powered usb hub and check if the camera works fine then.

How is it about your logging results?
Is there any output you may provide to us?

Best Regards
Kevin

TIS-Stefan, 2024-02-08:
Hello

> And then for starting and stopping the camera I go from Ready => Paused => Playing then Playing => Paused => Ready.

Going into pause state explicitly is not necessary in this situation, therefore, you may remove that from your code. 

If the camera is in trigger mode, it can happen, that starting the pipeline fails, because all modules may need a frame to prepare themself. Thus a state change to playing is never achieved. I suggest to disable trigger mode, start the stream and if the stream is in state playing, enable the trigger mode.  Same goes for stopping the stream. Disable the trigger mode, then stop the stream.  I see, you did that for starting the stream already. 

I do not think, it is a power issue. If so, you saw a four color icon in the upper right corner on the screen. 

Stefan

teawizardry, 2024-02-19:
I will try that and see if it fixes the issue. I'm unsure why but the gst log is not actually logging anything.

TIS-Stefan, 2024-02-20:
Did you run
export GST_DEBUG=:3,tcam:5
before starting your script? In the same terminal?
Stefan

closed 2024-11-28 by TIS-Edgar