TheImagingSource/tiscamera #544: Unable to use v4l2-ctl to update gain

xiliugatech, 2024-02-22:
Hi Stefan,

I am using 2 of the Imaging Source Europe GmbH DMK 37AUX290 for my project. Due to software incompatibility we are unable to import the tiscamera to our code base. So I tried to use linux v4l2 to control this camera.

We don't do any fancy things, just manually set the gain and exposure for the camera. I thought that should be pretty easy. However, turns out that using  v4l2-ctl --list-ctrls returns:
```
User Controls

               brightness 0x00980900 (int)    : min=0 max=511 step=1 default=240 value=240
                     gain 0x00980913 (int)    : min=0 max=240 step=1 default=0 value=239
Camera Controls: only 2 registers.

              auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=1 (Manual Mode)
     exposure_time_absolute 0x009a0902 (int)    : min=1 max=300000 step=1 default=3 value=280
```

When I tried to set the gain register using both c++ or v4l2-ctl, it won't allow me to do so. Looks like there's some auto gain setting that prevents manually adjusting the gain settings.

Earlier this year we tried to use the same cameras on another desktop computer but found the programability is much better: v4l2-ctl --list-ctrls returns:

```
User Controls

                     brightness 0x00980900 (int)    : min=0 max=511 step=1 default=240 value=240
                           gain 0x00980913 (int)    : min=0 max=240 step=1 default=0 value=166

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=1 (Manual Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=300000 step=1 default=3 value=300
               exposure_time_us 0x0199e201 (int)    : min=20 max=30000000 step=1 default=333 value=30000
                   auto_shutter 0x0199e202 (bool)   : default=1 value=0
        auto_exposure_reference 0x0199e203 (int)    : min=0 max=255 step=1 default=128 value=128
                      gain_auto 0x0199e205 (bool)   : default=1 value=1
                   trigger_mode 0x0199e208 (bool)   : default=0 value=0
               software_trigger 0x0199e209 (button) : value=0 flags=write-only
                  strobe_enable 0x0199e211 (bool)   : default=0 value=0
                strobe_polarity 0x0199e212 (bool)   : default=0 value=0
               strobe_operation 0x0199e213 (menu)   : min=0 max=1 default=1 value=1 (Exposure)
                strobe_duration 0x0199e214 (int)    : min=0 max=32767 step=1 default=100 value=100
                   strobe_delay 0x0199e215 (int)    : min=0 max=32767 step=1 default=0 value=0
                          gpout 0x0199e216 (bool)   : default=0 value=0
                           gpin 0x0199e217 (bool)   : default=0 value=0
                   roi_offset_x 0x0199e218 (int)    : min=0 max=1552 step=2 default=0 value=0
                   roi_offset_y 0x0199e219 (int)    : min=0 max=824 step=2 default=0 value=0
                roi_auto_center 0x0199e220 (bool)   : default=1 value=1
               trigger_polarity 0x0199e234 (menu)   : min=0 max=1 default=0 value=0 (Rising Edge)
     auto_functions_roi_control 0x0199e241 (bool)   : default=1 value=1
                flip_horizontal 0x0199e251 (bool)   : default=0 value=0
                  flip_vertical 0x0199e252 (bool)   : default=0 value=0
            highlight_reduction 0x0199e253 (bool)   : default=0 value=0
 exposure_auto_upper_limit_auto 0x0199e254 (bool)   : default=1 value=1
   exposure_auto_lower_limit_us 0x0199e255 (int)    : min=20 max=1000000 step=1 default=333 value=60
   exposure_auto_upper_limit_us 0x0199e256 (int)    : min=20 max=1000000 step=1 default=333 value=33333
         override_scanning_mode 0x0199e257 (int)    : min=1 max=16 step=1 default=1 value=0
      auto_functions_roi_preset 0x0199e258 (menu)   : min=0 max=1 default=1 value=1 (Center 50%)
          gain_auto_lower_limit 0x0199e259 (int)    : min=0 max=240 step=1 default=0 value=0
          gain_auto_upper_limit 0x0199e260 (int)    : min=0 max=240 step=1 default=240 value=240
   trigger_global_reset_release 0x0199e261 (bool)   : default=0 value=0
         scanning_mode_selector 0x0199e263 (int)    : min=1 max=16 step=1 default=1 value=1
       scanning_mode_identifier 0x0199e264 (int)    : min=1 max=1 step=1 default=1 value=1 flags=read-only
 scanning_mode_scale_horizontal 0x0199e265 (int)    : min=1 max=1 step=1 default=1 value=1 flags=read-only
   scanning_mode_scale_vertical 0x0199e266 (int)    : min=1 max=1 step=1 default=1 value=1 flags=read-only
        scanning_mode_binning_h 0x0199e267 (int)    : min=1 max=1 step=1 default=1 value=1 flags=read-only
        scanning_mode_binning_v 0x0199e268 (int)    : min=1 max=1 step=1 default=1 value=1 flags=read-only
       scanning_mode_skipping_h 0x0199e269 (int)    : min=1 max=1 step=1 default=1 value=1 flags=read-only
       scanning_mode_skipping_v 0x0199e270 (int)    : min=1 max=1 step=1 default=1 value=1 flags=read-only
            scanning_mode_flags 0x0199e271 (int)    : min=0 max=0 step=0 default=0 value=0 flags=read-only
            trigger_delay_100ns 0x0199e272 (int)    : min=0 max=100000 step=1 default=0 value=31

```

I am curious to know why there's difference between one desktop to another, and whether there's a way to simply use v4l2 to control gain and exposure of the camera. Thanks.

xiliugatech, 2024-02-22:
Quick update: I tried to capture by writing a c++ program to capture image via v4l2. In the program I set gain = 8. However, I noticed the gain seems to bounce back to a certain value and stopped. So looks like there's some internal register (most likely gain_auto 0x0199e205 (bool))  that is controlling the gain and we are not able to program it via v4l2?

TIS-Stefan, 2024-02-23:
Hello 

the missing properties are not known by v4l2 by default. Thus we made some udev rules, which add these properties to v4l2. You can follow the article at https://wiki.theimagingsource.com/index.php/Using_USB_2.0_CCD_Cameras_in_Linux, even if it is for using one of our older camera models. The udev file contains the rules for the modern cameras too. 

As you can see in your listing, gain is expected to have a range from 0 to 240. I suppose, you will get this range after installing the udev rules.
The 0x0199e205 is for gain auto. If gain auto is enabled (true), then it always will override a gain (0x00980913 ) value you try to set. 

Stefan