TheImagingSource/tiscamera #547: AttributeError: 'GstTcamBin' object has no attribute 'get_device_serials_backend' while running the 00

zarathacia, 2024-04-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**computer used (please complete the following information):**
 - OS: [Ubuntu 18.04, Debian Testing]
 - Architecture/Platform: [amd64, armhf, arm64], [Desktop, NVidia Nano]
 - Camera  model you use
 - tiscamera version (tcam-ctrl --version, if possible)

**Additional context**
Add any other context about the problem here.

If your bug report has potentially sensitive information, please write us directly at support@theimagingsource.com .

TIS-Stefan, 2024-04-05:
Hello
The environment  TCAM_DISABLE_DEVICE_BLACKLIST=1 variable must be set in all sessions you have. I you set it only once in the terminal, is is not set in another. 

I am sure, there is a ways to set this variable once for your user account, but I am not sure, how to do that (it is Linux OS thing.)

Stefan

zarathacia, 2024-04-05:
Hi, 
yeah it doesn't really relate to my problem is with the error that comes when i run the example files. i get the traceback above.

TIS-Stefan, 2024-04-05:
Hello
I just saw, you use version 1.1.0, but we are on version 1.1.1 already. Therefore, can you please update and try again?

Stefan

TIS-Stefan, 2024-04-05:
Hello

did you do the  "export TCAM_DISABLE_DEVICE_BLACKLIST=1" set before running the sample? What does tcam-ctrl -l says? (It does the same as the Python script.

Stefan

zarathacia, 2024-04-05:
Hi it worked. thanks for your assistance.

closed 2024-04-05 by zarathacia