TheImagingSource/tiscamera #548: support 23.04, 23.10, and 24.04

cdelguercio, 2024-04-07:


TIS-Edgar, 2024-04-10:
tiscamera should be able to compile on all mentioned Ubuntu versions.
If you are encountering problems, please provide additional information as to what the issues are.
Ubuntu 23 should be able to use Ubuntu 18 packages. The only difference between 18/20 packages are in the libzip dependency. 
Ubuntu 24.04 LTS will receive precompiled packages once it has been officially released.

closed 2024-04-10 by TIS-Edgar

jarledo, 2024-09-02:
Any progress on this? The latest release 1.1.1 does not compile on 24.04. Pulling from master works, but it would be nice to use a release tarball instead. It would be even better if you also provided binary packages for Ubuntu 24.04, but that appears to have stopped after 20.04?

TIS-Edgar, 2024-09-09:
As stated above:
Ubuntu 23, 24 and future release will be supported and are supported.

The Ubuntu 18 packages are compatible 23 and 24.
The master branch currently only contains small bug fixes for the latest stable release, such as ensuring compilations.
It can be downloaded as zip here: https://github.com/TheImagingSource/tiscamera/archive/refs/heads/master.zip

Stable releases do not happen on a regular basis. I can currently not say with certainty when the next one will happen.