TheImagingSource/tiscamera #549: Performance Variation Across Ubuntu Versions and tiscamera Package Versions

yukiogawa21, 2024-04-11:
I'm currently utilizing a Jetson(arm64) device to run an application for visual inspection within a container environment. We've been experimenting with the tiscamera package, available for Ubuntu 18, 20, and 22. Upon conducting the same operations within our application, we observed a performance discrepancy; the package for Ubuntu 20 exhibited roughly a 10ms delay in processing compared to the packages for Ubuntu 18/22. Additionally, upgrading from version 1.0.0 to 1.1.0 of the tiscamera package also resulted in an approximate 10ms increase in processing speed. While we continue to investigate other potential factors, I'm curious if the differences in tiscamera package versions are known to cause variations in processing speeds. Furthermore, why is there a separate package specifically for Ubuntu 20 as opposed to a unified package for Ubuntu 18/22?

<ubuntu18/22>
https://dl-gui.theimagingsource.com/en_US/c1997009-6524-5785-b42c-4b19efdbaff3/

<ubuntu20>
https://dl-gui.theimagingsource.com/en_US/528febcf-1aee-5456-b30d-5ba73fe883bf/

yukiogawa21, 2024-04-12:
Hi,

I am currently specifying the l4t-ml image and switching Ubuntu versions within the container to test. The application uses a GigE camera. Thank you for the detailed explanation regarding the different speeds between Ubuntu 18 and Ubuntu 20. Currently, I cannot fully replicate the application in an Ubuntu 22 container. If I use the tiscamera package for Ubuntu 18/22 in the Ubuntu 22 container, given that it's the same package, is it correct to assume that the processing speed could be similar to what it was with Ubuntu 18?

TIS-Edgar, 2024-04-12:
You should be able to use the Ubuntu18 packages, yes.
We cannot guarantee the same performance as gstreamer libraries, etc will be different but from our code no performance differences should occur.

If you want to save additional disk space and reduce the image size it may be a good idea to create the debian packages yourself. That way you can remove libusb and other dependencies that are not required for GigE.
It would not be much but might be something of interest for your task

yukiogawa21, 2024-04-12:
Thank you very much. Your assistance has been incredibly helpful.

closed 2024-04-12 by yukiogawa21