TheImagingSource/tiscamera #553: Use dfk 33ux287 tisCam code to install library fail

Rweichen, 2024-05-13:
[image]
Missing module gi when running tiscam code
pip install PyGObject fail

[image]

TIS-Stefan, 2024-05-13:
Hello 
I searched in the internet for "no module named gi" and found 
https://stackoverflow.com/questions/71369726/no-module-named-gi
and many more. 
However, as far as I know, WSL has not access to the USB port, thus you wont be able to access the camera in WSL at all. (May be that was changed, but I do not know)

Stefan

Rweichen, 2024-05-14:
I run it directly on the window, not WSL.

> 你好， 我在網路上搜尋「沒有名為 gi 的模組」並找到了 https://stackoverflow.com/questions/71369726/no-module-named-gi 等等。 但是，據我所知，WSL 無法訪問 USB 端口，因此您根本無法訪問 WSL 中的相機。 （可能已經改變了，但我不知道）
> 
> 斯特凡

Rweichen, 2024-05-14:
> 好，我在網上搜尋“ 沒有 叫 gi 的模組”並找到了 https://stackoverflow.com/questions/71369726/no-module-named-gi 等等。端口，因此您根本無法訪問WSL中的功能。
> 
> 斯特凡

Because when I run the program, it displays like this. I wonder if I have installed less libraries?

Rweichen, 2024-05-18:
[image]

issue is module gi

> 非常抱歉，我的中文很差。
> 
> > 因為當我運行程式時，它顯示如下。我想知道我是否安裝了較少的庫？
> 
> 顯示像什麼？
> 
> 但是，我從未在 WSL 中運行過 tiscamera，因此我在這裡沒有太大幫助。

Rweichen, 2024-05-18:
[image]
issue is Namespace Tcam not available

Rweichen, 2024-05-20:
Is there any gstreamer that is compatible with Windows? I think they are all for Linux.
I'm wondering if Namespace Tcam not available is related to gstreamer?

> I am very sorry, but my Chinese is very bad.
> 
> > Because when I run the program, it displays like this. I wonder if I have installed less libraries?
> 
> Displayed Like what?
> 
> However, I never ran tiscamera in WSL, thus I am not of big help here.

TIS-Edgar, 2024-05-28:
GStreamer is available for windows.
To have the module Tcam available you need to have gobject-introspection available.

tiscamera very explicitly has **no** windows support.

If you want to develop for python I highly recommend the use of IC4 as the way forward.
You can find the documentation for IC4 python here: https://www.theimagingsource.com/en-us/documentation/ic4python/index.html