TheImagingSource/tiscamera #554: Stuck setting pipeline state to ready - pipeline.set_state(Gst.State.READY) due to TriggerMode

Niv-R, 2024-05-15:
**Describe the bug**
When "TriggerMode" is set to "On", restarting the pipeline gets stuck at pipeline.set_state(Gst.State.READY)

**To Reproduce**
1. Set "TriggerMode" to "On"
2. Attempt to start pipeline and set it to READY

**computer used (please complete the following information):**
 - OS: Ubuntu 22.04
 - Architecture/Platform:  amd64 Desktop
 - Camera: DMK 39GX265-Z20-TI
 - tiscamera version:
 	Tcam:	1.1.1_HEAD/ee45fcf_rev_1
	Aravis:	0.8_version_e977fa4
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc

**Additional context**
The pipeline string contains valve before the sink
`tcamsrc serial=... name=source ! video/x-raw,format=GRAY12p,width=1920,height=1080,framerate=8192/225 ! valve name=valve ! appsink name=sink
`

Changing "TriggerMode" to "Off" between Gst.State.READY and Gst.State.PLAYING helped as a workaround.

TIS-Stefan, 2024-05-15:
Hello

The "valve" wont become ready, if there are no images. That is the blocking part. 
It even can happen, if the "valve" is "closed", that follwing modules do not get ready. 

Stefan

Niv-R, 2024-05-20:
I've tried removing the "valve" from the pipeline, but the issue still reproduced.

TIS-Edgar, 2024-05-28:
What is your current pipeline?

Set the environment variable `GST_DEBUG=*:3,tcam*:5`.
This will give you a log of what the elements in your pipeline are doing and if images are received/pushed into the pipeline.

rurban, 2024-12-10:
I think you need to wire the HW trigger cable.

Else turn the trigger mode off, which is called "Software Trigger".

TIS-Stefan, 2024-12-10:
The trigger mode is called "Trigger Mode". 
"Software Trigger" is a command, which is used to send a trigger command by software to the camera.

closed 2024-12-10 by TIS-Stefan