TheImagingSource/tiscamera #555: Cannot verify installation in Jetson AGX Orin

tunahanertekin, 2024-05-31:
Hi,

I'm trying to run [DMK 37AUR0234](https://www.theimagingsource.com/en-us/product/industrial/37u/dmk37aur0234/) camera on my NVIDIA Jetson AGX Orin (arm64 / Ubuntu 22.04) using the directives in README. However I couldn't verify installation using `tcam-capture` or `gstreamer`.

I completed the steps in README successfully. When I run `tcam-capture` and enable "Auto Select", a window pops up saying:

```
Unable to open device. Please check gstreamer log 'tcam*:5' for more information.
```

Here are the screenshots:

[image]

[image]

When I try to verify installation using `gstreamer`, the logs say:

```
Setting pipeline to PAUSED ...
ERROR: from element /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source: Failed to find any device to open.
Additional debug info:
/home/robolaunch/Desktop/dev/tiscamera/src/gstreamer-1.0/tcamsrc/gsttcamsrc.cpp(249): open_source_element (): /GstPipeline:pipeline0/GstTcamBin:tcambin0/GstTcamSrc:tcambin-source
ERROR: pipeline doesn't want to preroll.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
```

I should state that I only followed the steps available in README for reproducing the issue. I can start installation from scratch if you need installation logs and share them if you require. Thanks in advance.

 - OS: Ubuntu 22.04
 - Architecture/Platform: arm64 / NVIDIA Jetson AGX Orin w/ Jetpack 6
 - Camera: DMK 37AUR0234
 - tiscamera version:

```
Versions: 
	Tcam:	1.1.1_master/7da8da19_rev_4154
	Aravis:	0.8_version_e977fa4
	Modules:	gst_aravis_arvu3v_v4l2_libusb_tools_doc
```

TIS-Kevin, 2024-05-31:
Dear tunahanertekin,
in the first image you sent is no camera shown.
Please make sure the camera is connected via USB 3.
Do not use USB Type C cable which is used for charging Smartphones. These cables often have Data lanes disabled or work seriously bad.

Is any USB Port damaged / disabled on your Hardware? Make sure and try other USB Ports.

Best Regards,
Kevin

tunahanertekin, 2024-05-31:
I updated the camera model, it's DFK 37AUR0234.

tunahanertekin, 2024-05-31:
FYI, I discovered that there are no ports such as `/dev/videoX` after I plug the USB in to Jetson AGX Orin. The same camera's port was available in x86 Ubuntu 22.04 as `/dev/videoX`.

TIS-Kevin, 2024-05-31:
Dear tunahanertekin,
the camera should be listed by following command:
`sudo dmesg`

another command for listing usb devices is
`lsusb -t`

can you find the camera in both commands?
You can also send the Output to us as textfile so we can check.

tunahanertekin, 2024-05-31:
Hi @TIS-Kevin,

Here are the outputs of the commands you requested. (and `v4l2` device list)

---

```bash
sudo dmesg
```
[dmesg-output.txt](https://github.com/user-attachments/files/15513460/dmesg-output.txt)

---

```bash
lsusb -t
```

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 480M
    |__ Port 3: Dev 2, If 0, Class=Wireless, Driver=rtk_btusb, 12M
    |__ Port 3: Dev 2, If 1, Class=Wireless, Driver=rtk_btusb, 12M
    |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 3: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

---

```bash
v4l2-ctl --list-devices
```

```
NVIDIA Tegra Video Input Device (platform:tegra-camrtc-ca):
	/dev/media0

Cannot open device /dev/video0, exiting.
```

---

ps. Here's the summary of latest situation :

- Camera (TIS DFK 37AUR0234) can work on these platforms and has a video port (`/dev/videoX`)
  - amd64 / PC / Ubuntu 22.04
  - arm64 / Orange Pi / Ubuntu 22.04
  - PC / Windows 10/11
- In Jetson AGX Orin (Jetpack 6 / Ubuntu 22.04)
  - No video port (`/dev/videoX`) when plugging TIS DFK 37AUR0234
  - Other USB cameras can work
    - Intel Realsense D435i is available as `/dev/video0`
    - Intel Realsense D435i can function

TIS-Kevin, 2024-06-06:
Dear tunahanertekin,
i tried a Jetson AGX Orin (Jetpack 5) and it worked like a charm.
Since you confirmed the camera is not broken via the amd64 PC i suggest to try JetPack 5 or get in contact with Nvidia Support.

Best Regards
Kevin