TheImagingSource/tiscamera #557: A solution for DMK 21BUC03?

quantumm, 2024-08-29:
Hi everyone,

First, thank you for your work to maintain an open-source package for Linux. If only all suppliers would do like you ;)

I have found a probably quite old model DMK 21BUC03 connected with USB. Do you have a solution for this model?

By the way, i am part of the team maintaining the [PyMoDAQ](https://pymodaq.cnrs.fr/en/latest/) project, which is an open-source framework to develop data acquisition programs. Could be interesting for you. Especially since we want to promote suppliers like you among the scientific community.

All the best!

TIS-Kevin, 2024-08-30:
Hello quantumm,
as you can see in our Supported Device List:
https://www.theimagingsource.com/en-us/documentation/tiscamera/supported-devices.html
the 21 Series is not supported.

A possible way to use the camera with tiscamera could be to disable the Blacklist. The camera is not supported thus the functionality is not guaranteed.
https://www.theimagingsource.com/en-us/documentation/tiscamera/environment.html#tcam-disable-device-blacklist

quantumm, 2024-08-30:
Ok I will have a look at that.

Thanks for your reactive response.

TIS-Edgar, 2024-09-09:
A bit further information:

Older cameras like 21,31,41,61 etc are not supported by tiscamera. Never have, never will be. They are just. to. old.

They are _in theory_ v4l2 compatible.
They do not notify the system that they are UVC devices, 
so you need to tell the system yourself.

You can do that with udev rules, documented here:
https://github.com/TheImagingSource/tiscamera/blob/7da8da1938f47a8d38d285175d92508f0de9533b/data/udev/80-theimagingsource-cameras.rules.in#L5-L28

Only the legacy section, nothing more.
If the video formats do not work properly then that is all the camera will deliver in that mode and you are on your own.

quantumm, 2024-09-10:
Thanks for all those informations. I will see what I can do.

Best

closed 2024-11-28 by TIS-Edgar