TheImagingSource/tiscamera #560: Error: Getting Blank Screen When Trying to capture image using v4l2

SwasthikaPoojary, 2024-10-08:
Hi,
I am trying to capture a image using a USB Camera(DMM72BUC02-ML Monochrome Board Camera)  in Jetson Orin Nano. While I am able to capture images using the Aravis , I am facing issues when attempting to capture images via V4L2. Despite following the required steps, I am unable to retrieve an image from the camera using V4L2. I'm getting a blank screen.

Could you assist in resolving this issue or provide guidance on troubleshooting the V4L2 image capture on Orin Nano?

SwasthikaPoojary, 2024-10-10:
Thanks for the reply!

I’ve Followed the exact steps you have provided:

git clone https://github.com/TheImagingSource/tiscamera.git
cd tiscamera
sudo ./scripts/dependency-manager install
mkdir build
cd build
cmake -DTCAM_BUILD_ARAVIS=OFF 
cmake -DTCAM_BUILD_V4L2=<ON>
make
sudo make install
Then I ran command: gst-launch-1.0 tcambin ! videoconvert ! ximagesink to capture image, but I’m getting a blank screen.

[image]

When I ran the tcam-capture  command, I get two options: Aravis and V4L2, I can capture image using Aravis and Unable to capture image using V4L2.

> Does a /dev/video* node exist for the camera? `ll /dev/v4l/by-id/` will show you associations between devices and nodes.
-->Yes, the node exists for camera.

> > If it exists, does qv4l2 maybe work? That is a v4l2 test bench. `sudo apt install qv4l2`
---> Will try it and let you know

TIS-Stefan, 2024-10-14:
Hello

you must use the cmake parameters in one command line:
```
cmake -DTCAM_BUILD_ARAVIS=OFF -DTCAM_BUILD_V4L2=ON ..
```
Did you run the Linux firmware update for your camera? If not, it is not v4l2 compatible. The firmware update can be cloned from https://github.com/TheImagingSource/tcam-firmware-update . The firmware to be written into your camera is [dmk72uc02_162_uvc.euvc](https://github.com/TheImagingSource/tcam-firmware-update/blob/master/firmware/usb2/dmk72uc02_162_uvc.euvc)

Stefan

closed 2026-01-21 by TIS-Edgar