TheImagingSource/tiscamera #565: unable to find tcamconvert element

kijsskidata, 2024-12-13:
Hello there,

I build the tiscamera gst-plugins for our systems. Using tis cameras with the tcamsrc element works just fine. But for auto white balance I tried using the tcamconvert gst element. Unfortunately, the element does not seem to being found by gstreamer. 

```
gst-inspect-1.0 tcamconvert
No such element or plugin 'tcamconvert'
```

I am not so used to debugging gstreamer plugins, so hints on how to debug the problem would be much appreciated.

Best regards Johannes

TIS-Edgar, 2024-12-13:
Hi there,
To see if tcamconvert is in you system:

```find /usr/lib -name "*tcamconvert*"```

it should be in the same directory as libgsttcamsrc.so. On amd64 systems `/usr/lib/x86_64-linux-gnu/gstreamer-1.0/`.
If it is in that folder:
```gst-inspect <absolute path to libgsttcamconvert.so>```

That will tell you why that shared library cannot be loaded.

If you cannot find the so file, something went wrong beforehand and you will have to tell me how you configured/installed tiscamera.

kijsskidata, 2024-12-13:
Hey,

It is here on my system:
```
# ll /usr/lib/gstreamer-1.0/*tcam*
-rwxr-xr-x 1 root root  63816 Nov  9  2024 /usr/lib/gstreamer-1.0/libgstcamerabin.so
-rwxr-xr-x 1 root root 380744 Dec 13  2024 /usr/lib/gstreamer-1.0/libgsttcambin.so
-rwxr-xr-x 1 root root 146712 Jan  1 15:26 /usr/lib/gstreamer-1.0/libgsttcamconvert.so
-rwxr-xr-x 1 root root 471400 Dec 13  2024 /usr/lib/gstreamer-1.0/libgsttcamsrc.so
-rwxr-xr-x 1 root root 146720 Dec 13  2024 /usr/lib/gstreamer-1.0/libtcamconvert.so
```
It misses the gst prefix, but I guess this is due to the Cmakefile https://github.com/TheImagingSource/tiscamera/blob/7da8da1938f47a8d38d285175d92508f0de9533b/src/gstreamer-1.0/tcamconvert/CMakeLists.txt#L16 , but not sure.

Anyways, Inspect tells me: 
```
~# gst-inspect-1.0 /usr/lib/gstreamer-1.0/libtcamconvert.so > out.txt
Illegal instruction
~# echo out.txt  
out.txt
```

TIS-Edgar, 2024-12-13:
It misses the architecture specific subfolder. That happens on some systems. Nothing to worry about.
We ask gstreamer pkg-config where the elements should be installed to. You can find that [here](https://github.com/TheImagingSource/tiscamera/blob/master/CMakeInstall.cmake#L147).

To print the content of a file in bash execute `cat out.txt`. 
`echo` will only print the text `out.txt` if you use it like that.

kijsskidata, 2024-12-13:
Ups, my bad.
The file is empty anyways.
```
~# cat out.txt 
~#
```

To me it seems like the kernel kills the process because of an "Illegal instruction". (maybe some none aligned read or something, arm is picky with those). 
Will build gdb and install debug infos, maybe we find the reason by doing this.
But thank you already for telling me that one can use ```gst-inspect <path>``` (Did not know that)

kijsskidata, 2024-12-18:
Found the time to have the debugger run. It seems like the tcamconvert element uses neon functions if it is compiled for ARM, e.g. in transoform_impl.c
Our older products use ARM CPUs without NEON, will have a look if I can fix this easily.

kijsskidata, 2024-12-19:
Added the pullrequest here https://github.com/TheImagingSource/tiscamera/pull/566. It's probably far from a perfect sollution. So please have a look.